RinCat/RTL88x2BU-Linux-Driver #2: Monitor mode support

xavier83, 2019-04-16:
Does this driver have monitor mode fixes for the issue described at https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959/issues/9# ?

RinCat, 2019-04-16:
I don't think it is fixed. This one is the original driver with some kernel compatibility patches.

arkhub, 2020-03-25:
sorry to ressurect old issues, is there currently driver which support monitor mode?

RinCat, 2020-04-07:
Not yet.

RinCat, 2020-09-20:
You guys can give it a try.

j4munoz, 2020-10-26:
I can confirm that Monitor mode is working on RTL8812BU chipset.
Injection isn't, but I already forked and managed to make it work for injection too. I'll create a PR.

xavier83, 2020-12-01:
Does https://github.com/RinCat/RTL88x2BU-Linux-Driver/pull/51 close this issue @j4munoz @RinCat  ?

j4munoz, 2020-12-05:
Hi @xavier83 , my PR did not modify the monitor mode behavior, when I tried monitor mode on the RTL8812BU it was working fine, nothing to do there. Injection was only working for frames smaller than a particular number of bytes. Now it should work for any frame length.
I don't know if monitor mode is working also in other chipset models sorry.