RinCat/RTL88x2BU-Linux-Driver #3: Error?

vav4eg, 2019-05-01:
Hi, when I install the driver, the following error appears. But the driver is assembled and installed. How critical is it?

make[1]: вход в каталог «/usr/src/linux-headers-5.0.10-050010-generic»
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_cmd.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_security.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_debug.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_io.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ioctl_query.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ioctl_set.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ieee80211.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mlme.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mlme_ext.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mi.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_wlan_util.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_vht.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_pwrctrl.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_recv.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_sta_mgt.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ap.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_xmit.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_p2p.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rson.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_tdls.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_br_ext.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_iol.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_sreset.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_btcoex_wifionly.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_btcoex.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_beamforming.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_odm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rm_fsm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/efuse/rtw_efuse.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/osdep_service.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/os_intfs.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/usb_intf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/xmit_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/mlme_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/recv_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/wifi_regd.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_android.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_proc.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.o
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_reg’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:46:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_read_reg’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:132:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_rf’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:266:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_read_rf’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:308:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_rate’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:417:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
[... 155 lines cut ...]
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/platform/platform_ops.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mp.o
  LD [M]  /home/vav4eg/RTL88x2BU_1/88x2bu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/vav4eg/RTL88x2BU_1/88x2bu.mod.o
  LD [M]  /home/vav4eg/RTL88x2BU_1/88x2bu.ko
make[1]: выход из каталога «/usr/src/linux-headers-5.0.10-050010-generic»

RinCat, 2019-05-01:
Seems your gcc still use outdated ISO C90.
Try build with 
`CFLAGS="-std=c11" make`

vav4eg, 2019-05-01:
I'm sorry, I'm new to this system, tell me, and how to upgrade?) - gcc still use outdated ISO C90

**Conclusion on the advice (if I did everything correctly)**

root@ubuntu:/home/vav4eg/RTL88x2BU_1# CFLAGS="-std=c11" make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.0.10-050010-generic/build M=/home/vav4eg/RTL88x2BU_1  modules
make[1]: вход в каталог «/usr/src/linux-headers-5.0.10-050010-generic»
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_cmd.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_security.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_debug.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_io.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ioctl_query.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ioctl_set.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ieee80211.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mlme.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mlme_ext.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mi.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_wlan_util.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_vht.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_pwrctrl.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_recv.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_sta_mgt.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_ap.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_xmit.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_p2p.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rson.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_tdls.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_br_ext.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_iol.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_sreset.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_btcoex_wifionly.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_btcoex.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_beamforming.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_odm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_rm_fsm.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/efuse/rtw_efuse.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/osdep_service.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/os_intfs.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/usb_intf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/xmit_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/mlme_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/recv_linux.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/wifi_regd.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_android.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_proc.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.o
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_reg’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:46:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_read_reg’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:132:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length + 1];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_write_rf’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:266:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
  char input[wrqu->length];
  ^~~~
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c: In function ‘rtw_mp_read_rf’:
/home/vav4eg/RTL88x2BU_1/os_dep/linux/ioctl_mp.c:308:2: warning: ISO C90 forbids variable length array ‘input’ [-Wvla]
[... 159 lines cut ...]
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/rtl8822b/phydm_rtl8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/phydm/txbf/haltxbf8822b.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822bwifionly.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822b1ant.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/hal/btc/halbtc8822b2ant.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/platform/platform_ops.o
  CC [M]  /home/vav4eg/RTL88x2BU_1/core/rtw_mp.o
  LD [M]  /home/vav4eg/RTL88x2BU_1/88x2bu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/vav4eg/RTL88x2BU_1/88x2bu.mod.o
  LD [M]  /home/vav4eg/RTL88x2BU_1/88x2bu.ko
make[1]: выход из каталога «/usr/src/linux-headers-5.0.10-050010-generic»

RinCat, 2019-05-01:
Which system you are using? and version?

vav4eg, 2019-05-01:
Ubuntu 18.04.2 LTS
Linux ubuntu 5.0.10-050010-generic #201904270832 SMP Sat Apr 27 08:34:43 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
5.0.10-050010-generic

vav4eg, 2019-05-01:
Linux version 5.0.10-050010-generic (kernel@kathleen) (gcc version 8.3.0 (Ubuntu 8.3.0-6ubuntu1)) #201904270832 SMP Sat Apr 27 08:34:43 UTC 2019

RinCat, 2019-05-01:
Seems the warning can be ignored.

vav4eg, 2019-05-01:
Thank you very much for the answer and advice :-)

RinCat, 2019-05-01:
Some of code using newer C standard and the old C not support, but gcc seems can deal with it. So no affect.

vav4eg, 2019-05-01:
Thank you very much for your code, I have been trying various versions for a very long time, but only your code works for me

RinCat, 2020-08-02:
I spent some time fixed all warnings in the new version. At least no longer have any warning for me in gcc 9.3 and clang 10.

closed 2020-09-20 by RinCat