RinCat/RTL88x2BU-Linux-Driver #4: Please add supported Kernel 5.1

vav4eg, 2019-05-06:
root@ubuntu:/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver# make clean
#make -C /lib/modules/5.1.0-050100-generic/build M=/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
root@ubuntu:/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.1.0-050100-generic/build M=/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver  modules
make[1]: вход в каталог «/usr/src/linux-headers-5.1.0-050100-generic»
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_security.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_debug.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_io.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_ioctl_query.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_ioctl_set.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_ieee80211.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_mlme.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_mlme_ext.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_mi.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_wlan_util.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_vht.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_pwrctrl.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_rf.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_recv.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_sta_mgt.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_ap.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_xmit.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_p2p.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_rson.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_tdls.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_br_ext.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_iol.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_sreset.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_btcoex_wifionly.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_btcoex.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_beamforming.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_odm.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_rm.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/rtw_rm_fsm.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/core/efuse/rtw_efuse.o
  CC [M]  /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/os_dep/osdep_service.o
/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/os_dep/osdep_service.c: In function ‘isFileReadable’:
/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/os_dep/osdep_service.c:2190:10: error: implicit declaration of function ‘get_ds’; did you mean ‘get_da’? [-Werror=implicit-function-declaration]
   set_fs(get_ds());
          ^~~~~~
          get_da
/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/os_dep/osdep_service.c:2190:10: error: incompatible type for argument 1 of ‘set_fs’
In file included from ./include/linux/uaccess.h:11:0,
                 from ./include/linux/crypto.h:26,
                 from ./include/crypto/hash.h:16,
                 from ./include/linux/uio.h:14,
                 from ./include/linux/socket.h:8,
                 from ./include/linux/compat.h:15,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/include/osdep_service_linux.h:30,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/include/osdep_service.h:50,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/include/drv_types.h:27,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/os_dep/osdep_service.c:19:
[... 25 lines cut ...]
           ^~~~~~
In file included from ./include/linux/uaccess.h:11:0,
                 from ./include/linux/crypto.h:26,
                 from ./include/crypto/hash.h:16,
                 from ./include/linux/uio.h:14,
                 from ./include/linux/socket.h:8,
                 from ./include/linux/compat.h:15,
                 from ./include/linux/ethtool.h:17,
                 from ./include/linux/netdevice.h:41,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/include/osdep_service_linux.h:30,
                 from /home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver/include/osdep_service.h:50,
[... cut ...]
Makefile:1571: recipe for target '_module_/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver' failed
make[1]: *** [_module_/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver] Error 2
make[1]: выход из каталога «/usr/src/linux-headers-5.1.0-050100-generic»
Makefile:1986: recipe for target 'modules' failed
make: *** [modules] Error 2
root@ubuntu:/home/vav4eg/wifi/rtl88x2BU_WiFi_linux_v5.2.4.4_26334.20180126_COEX20171012-5044/RTL88x2BU-Linux-Driver#

RinCat, 2019-05-06:
WIP

RinCat, 2019-05-06:
Try again? I just updated the code.

vav4eg, 2019-05-06:
Thank you so much. It turned out to collect and the device works

closed 2019-05-06 by RinCat