RinCat/RTL88x2BU-Linux-Driver #5: Help

vav4eg, 2019-05-09:
Good afternoon, I am trying to assemble a driver on another computer, but it will not work for me. Can you advise something?

Linux note 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64 GNU/Linux

 CC [M]  /home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.o
/home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.c: In function ‘dump_drv_version’:
/home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
/home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.c:45:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
/home/bender/T4Uv3_WiFi_linux_v5.3.1_beta/RTL88x2BU-Linux-Driver/core/rtw_debug.c:45:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
cc1: some warnings being treated as errors

RinCat, 2019-05-09:
Check the Makefile line 18-21, it has a workaround, not sure why not work in your system.
If the auto check failed, remove form `GCC_VER_49` to `endif`, keep `EXTRA_CFLAGS += -Wno-date-time`.

vav4eg, 2019-05-09:
Thank you very much! You have once again solved my problem!

closed 2019-05-10 by RinCat