RinCat/RTL88x2BU-Linux-Driver #6: Kernel 5.2/5.3

nukedathlonman, 2019-07-27:
Any idea what needs to tweaked to allow it to be compiled on newer/bleeding edge kernels (like Kernel 5.2 and 5.3)?  I wanted to test some newer kernels, but I'm unable to compile the module for use in them.

RinCat, 2019-08-06:
This one should works in 5.2, didn't test in 5.3, if you have error log please post here.

nukedathlonman, 2019-08-09:
Build log for Kernel 5.3 RC3 attached.  I get the exact same error with 5.2.x  Kernels are from mainline (I'm using the Kubuntu 19.04).  I don't have any problems with any 5.1.x or older kernels.  I do have this configured up with dkms for some automation when changing kernels.

Log:
DKMS make.log for rtl88x2bu-5.3.1 for kernel 5.3.0-050300rc3-generic (x86_64)
Mon Aug  5 12:24:36 MDT 2019
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-050300rc3-generic/build M=/var/lib/dkms/rtl88x2bu/5.3.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-050300rc3-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.o
/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.c:1828:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
    ptable->func = &OnAuthClient;
    ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mlme_ext.c:1830:2: note: here
  case WIFI_ASSOCREQ:
  ^~~~
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.o
/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.c:1433:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.c:1433:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:274: /var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1621: _module_/var/lib/dkms/rtl88x2bu/5.3.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-050300rc3-generic'
make: *** [Makefile:1986: modules] Error 2

RinCat, 2019-08-09:
Did you use the latest git version?

nukedathlonman, 2019-08-09:
Yes, I did a fresh download just in case.

RinCat, 2019-08-09:
Show me this result:
`sed -n '1300,1305p' "/var/lib/dkms/rtl88x2bu/5.3.1/build/os_dep/linux/os_intfs.c"`

nukedathlonman, 2019-08-09:
Ack...I see the problem.  I so don't get git - I thought I had done the pull properly.  I tried a few more times, nope.  So I just downloaded the entire thing in zip format and unzipped it to the my git directory - all is good now.

closed 2019-09-11 by RinCat