RinCat/RTL88x2BU-Linux-Driver #7: support for FTM – possibly only requires changing a flag?

codingcatgirl, 2019-10-23:
Hey,

i'm trying to use my RTL8812BU WiFi USB Adapter to set up an AP with Wi-Fi RTT support using hostapd. According to various sources, like [this one](https://www.netspotapp.com/what-is-wifi-rtt.html), the chip is certified to support this. However, if i configure hostapd to advertise this feature, i get the message: `Not configuring FTM responder as the driver doesn't advertise support for it`.

I'm using your driver and i wonder if this could be fixed very easily? Looking at the hostapd source code, it just looks at some driver support flags and that's it. It also says in the documentation: `This parameter only controls publishing via Extended Capabilities element. Actual functionality is managed outside hostapd.`. Since the actual RTT logic should be happening on the chip, i wonder if maybe all you need to do is advertise support for this feature so that hostapd can use it? If i understand correctly, this is just about changing the beacon data.

Regardless, even if it wasn't as easy as just advertising support for it, it would be wonderful if Wi-Fi RTT was supported by this triver since using USB dongles with this chip is more or less the only cheap way right now to get a device that (theoretically) can be used as a Wi-Fi RTT responder.

RinCat, 2019-10-23:
I just checked the code, but sadly the driver not support it.
It do check if the hardware support FTM, but nothing else will do. 
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/core/efuse/rtw_efuse.c at line 3010
It could be only Windows driver support FTM or the opensource driver is too old. (The newest I can found is about 1 year ago).

georgcampana, 2020-08-24:
> Hey,
>
> i'm trying to use my RTL8812BU WiFi USB Adapter to set up an AP with Wi-Fi RTT support using hostapd. According to various sources, like [this [...]

Hi CodingCatGirl.  I'm also trying to enable FTM on that USB device. There is indeed some disabled RTT code in the 5.6.1 driver but if you try to enable it, it's then missing other functions which are not present and apparently coming from the Android world.
To recap: the HW definitely supports FTM but there are no public drivers using this capability.
Did you find any way ? Or maybe a suitable linux driver that has the RTT part enabled ?

codingcatgirl, 2020-08-25:
Hi. Sadly i was unable to find a solution and have given up on it for now.