RinCat/RTL88x2BU-Linux-Driver #8: Short disconnects every few seconds

witchent, 2019-11-16:
Hey there,
I am using this driver for my usb wifi dongle. It works fine connecting to networks, but it is slower as expected and it disconnects/goes to sleep every few seconds as shown using dmesg:

> [ 1661.753261] RTW: cfg80211_rtw_scan(wlp9s0f3u1i2)
> [ 1661.759980] RTW: [HW_VAR_CHECK_TXBUF] Empty in 0 ms
> [ 1661.760145] RTW: wlp9s0f3u1i2 sleep m0=0x00000003, ori reg_0x4d4=0x00000000
> [ 1666.130769] RTW: wlp9s0f3u1i2 wakeup m0=0x00000003, ori reg_0x4d4=0x00000003
> [ 1666.131883] RTW: survey done event(2f) band:0 for wlp9s0f3u1i2
> [ 1666.131957] RTW: rtw_indicate_scan_done(wlp9s0f3u1i2)

This repeats every few seconds and in those 5 seconds I have no internet connection (tested with ping 8.8.8.8).

lsusb -v gives this related output:
Bus 003 Device 002: ID 0bda:b82c Realtek Semiconductor Corp. 802.11ac NIC

I have created the file /etc/modprobe.d/88x2bu and entered:
`options 88x2bu rtw_power_mgnt=0 rtw_ips_mode=0 rtw_enusbss=0`

iwconfig shows 
>Power Management:off

And finally I set a fixed BSSID in NetworkManager.
Nothing has worked so far. Do you have any more ideas what could help here? This is really making the stick unusable.

Thanks a lot

RinCat, 2019-11-17:
Hi, when did you starts using it? After yesterday update or always? Thanks.

witchent, 2019-11-17:
I am not sure to be honest. It had some bad moments before but I think it being that it only started to get that bad one or two days ago. Though as I am using Arch that might have been an update unrelated to this one as well.
I will try if I am able to compile and install the version from before yesterdays update and report back if that changes anything. Thanks a lot already

RinCat, 2019-11-17:
issue_addba_rsp_wait_ack with status 0 means 802.11 block acknowledgment ACK failed.
Have you tired try connect different wifi?

witchent, 2019-11-17:
Yes, I tried 3 different wifi, all behave in the same way. Also the stick works under windows, and using my phone with tethering, archlinux works with those wifis as well, so I don't think it's the router.
Any other ideas? I tried compiling the driver with edited makefile (CONFIG_POWER_SAVING = y -> n), but that didn't help either.

RinCat, 2019-11-17:
What the brand and model?

witchent, 2019-11-17:
https://www.amazon.com/dp/B07X3HNNG8/

So Yehua is the brand, no idea what that model is called though

witchent, 2019-11-17:
Something new I found out:
In Windows the stick works in every port, and as I just noticed, it is fast in all ports but the one I am using under linux. 
If I try to use any other port in linux though it fails with "configuring interface" and dmesg goes crazy. I uploaded the log on pastebin as it is quite lengthy:
https://pastebin.com/QvbakHYC

Maybe if you can help me get the dongle work in other ports that would solve the other issue as well. Any ideas what I could try to get it to work?

RinCat, 2019-11-17:
Weird errors and warnings, try remove your options for 88x2bu and check if anything changed.

RinCat, 2019-11-18:
It feels more like a power / USB compatibility issues rather than driver issues. As other user and I don't have those scan timeout and `rtw_ips_pwr` related logs. Do you set some aggressive power-saving mode?

witchent, 2019-11-18:
I did remove the options again (in fact I only added them after I was not able to get it to work properly) and it changed nothing.
For Power/USB compatibility issue, I have absolutely no idea how to go about checking that, because the ports work fine for my usb sticks in linux and for the dongle in windows, but the combination linux and wifi dongle only seems to work (with those annoying hickups) in one specific port.
As it is a desktop pc I have no power-saving modes activated (at least to my knowledge, never fiddled with anything in regards to that after installing barebone arch+kde). Any specific settings/programs you might have in mind that I could check?

Thank you so much for trying to help me again.

RinCat, 2019-11-18:
Try `lsusb` and get 
`Bus A Device B:  XXXXXX`
then `lsusb -v -s A:B` to see USB details.

All your USB ports should have the same issue if the bug is in the driver. But since it "works" at one specific port, it more likely incompatible.

witchent, 2019-11-18:
As far as I can see it shows the same description in every port. Anyway, I will probably just buy another wifi stick then and check if it works with that one. Thanks a lot for trying to help and maintaining this driver.

RinCat, 2019-11-18:
Can I see the result? Especially `bcdUSB` and `MaxPower`.

witchent, 2019-11-18:
Sure. The first one is what nearly every single usb port gives me (I tested them all now), where one of them is working and all other are not:
https://pastebin.com/sQTtA4nx
They are all exactly the same except for the bus and device obviously.

There is only one port with a different output (only bInterval is 1 instead of 4 and wMaxPacketSize is different, everything else is the same as well):
https://pastebin.com/sYSgpSJZ
This one does not work either, but I thought it might be interesting nonetheless.

zuzia-dev, 2019-11-28:
I have a similar problem too.

Port USB 3.0.
```
modprobe 88x2bu rtw_power_mgnt=0 rtw_ips_mode=0 rtw_enusbss=0 rtw_switch_usb_mode=1
lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 5: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
```

```
lsusb -v
Bus 004 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 
  bcdDevice            3.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
 
    **MaxPower              126mA**

      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
```
And disconnects every few seconds.

Port USB 2.0.
```
modprobe 88x2bu rtw_power_mgnt=0 rtw_ips_mode=0 rtw_enusbss=0 
lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 5: Dev 7, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
    |__ Port 6: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

lsusb -v
Bus 003 Device 007: ID 0bda:b812 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC

    **MaxPower              500mA**

  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
```
It works fine.

RinCat, 2019-11-28:
@witchent can you post a full `lsusb -v -s ID` log?

witchent, 2019-11-28:
I think you meant to tag @zuzia-dev , right? I already did.

RinCat, 2019-11-29:
@witchent yep, sorry.

zuzia-dev, 2019-11-29:
-   [**lsusb**](https://pastebin.com/Gm4A9W7F)

- [**dmesg**](https://pastebin.com/JCbeSNfx)

RinCat, 2020-08-02:
Updated to a new version, maybe you guys can try it again if you still have it.

witchent, 2020-08-06:
I just tried it, it seems to be working on the one port, no sudden disconnects so far. 
All other ports are still not working for some reason, but I don't mind that. So feel free to close the issue.

RinCat, 2022-05-21:
@datio Hi, maybe try different USB ports. Many reasons can cause this, and I don't really have a good way to debug it.

julian-zatloukal, 2023-04-29:
Similar issue, just like @datio. Disconnects on USB 2.0 and USB 3.0. Really annoying to be honest. I'll try different ports.

EuroV8, 2023-06-15:
Same issue as @datio, same exact dongle (TechKey), switching ports only seems to work momentarily