RinCat/RTL88x2BU-Linux-Driver #9: Support for Netgear A6150

sphadley, 2019-11-22:
rtl8812BU
0846:9055 NetGear, Inc. A6150

Kernel seems to load ok, but I don't see the network adapter show up in ip link. I am using the arch linux aur package.  I suspect maybe I am failing the whitelist as suggested in the readme page?

RinCat, 2019-11-22:
Try again. But I did not found anything official said that this chip is 8812/8822.

sphadley, 2019-11-22:
It didn't work, though looking at the code it looks like you added it as 8822.  This page as well as several forum posts that link to you seem to indicate to me that it should be 8812 http://en.techinfodepot.shoutwiki.com/wiki/Netgear_A6150. To be fair I'm not sure I know the difference.

Also now in my dkms status I get: rtl88x2bu, 5.6.1.r44.0891f5e, 5.3.6-arch1-1-ARCH, x86_64: installed (WARNING! Diff between built and installed module!)

I'm not sure how to resolve that and concerned maybe I didn't uninstall all the way?

Please advise.

RinCat, 2019-11-22:
Because 8812B and 8822B use the same driver.
Try remove the package and install again? I may need unload the module first.

sphadley, 2019-11-22:
I figured out how to do dkms force install, and with your update it started working! Thanks for the help and quick response!

RinCat, 2019-11-22:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/0891f5efc0a0df12d4e10d408ca9e94a4406a73b

closed 2019-11-22 by RinCat