RinCat/RTL88x2BU-Linux-Driver #10: Linxu 5.4 breaks this driver

kth5, 2019-11-27:
It seems the kernel release of 5.4 broke this driver during buildtime already:

```
  LD [M]  /home/kth5/code/own/RTL88x2BU-Linux-Driver/88x2bu.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "proc_get_parent_data" [/home/kth5/code/own/RTL88x2BU-Linux-Driver/88x2bu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1609: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.4.0-arch1-1-ARCH/build'
make: *** [Makefile:2245: modules] Error 2
```

This is the result trying to build 0891f5efc0a0df12d4e10d408ca9e94a4406a73b on powerpc64le with the following make call:

```
make ARCH=powerpc
```

RinCat, 2019-11-27:
Did the driver works before? Since it only said support x86/ARM/MIPS.

RinCat, 2019-11-27:
It defined at https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/os_dep/linux/rtw_proc.c#L39 ,not sure why it will fail.

RinCat, 2019-11-27:
It also defined in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/linux/proc_fs.h , which should works in current version kernel.

kth5, 2019-11-28:
Yes, it's been working fine with 5.2.11 up until 5.3.13, set aside that there were a few very rare connectivity performance issues.

The step that seems to fail is modpost, the kernel symbols seem to get exported differently:

[image]

There is no KCONFIG I'm aware off that would change the export of this particular function.

RinCat, 2019-11-28:
It seems your 5.4 kernel System.map missed something. Try build again?
My current 5.4 kernel symbols, and the driver works fine.
```
./linux-5.4.0-gentoo/System.map:ffffffff812b9250 T proc_get_parent_data
./linux-5.4.0-gentoo/System.map:ffffffff81ef73b4 r __ksymtab_proc_get_parent_data
./linux-5.4.0-gentoo/System.map:ffffffff81f0f215 r __kstrtab_proc_get_parent_data

./linux-5.4.0-gentoo/Module.symvers:0xf65086b8  proc_get_parent_data            vmlinux EXPORT_SYMBOL_GPL
```

kth5, 2019-11-28:
Hmm, now that is odd. Would you be willing to share your .config?

RinCat, 2019-11-28:
It just default config with some tweaks and personal preferences. I didn't change the kernel module / map neither. Maybe you can try the default configuration and see if it works.

kth5, 2019-11-28:
I'll do just that and report back once I found the option responsible for my woes. :)

kth5, 2019-12-09:
Ok, so I got around to rebuilding my kernel and a new config did fix the issue.

It seems an oldconfig on a Linux 5.3 .config caused this.

closed 2019-12-09 by kth5