RinCat/RTL88x2BU-Linux-Driver #11: Archer T4U v3 (TP-Link AC1300) not detected

Noisycall, 2020-01-16:
### lsusb result

Bus 002 Device 002: ID 1058:25e2 Western Digital Technologies, Inc. My Passport (WD40NMZW)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 004: ID 04d9:1829 Holtek Semiconductor, Inc. Gaming Keyboard
**Bus 001 Device 012: ID 2357:0115 TP-Link 802.11ac NIC**
Bus 001 Device 002: ID 046d:c07e Logitech, Inc. G402 Gaming Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### lshw result
*-usb:1 UNCLAIMED
                   description: Generic USB device
                   product: 802.11ac NIC
                   vendor: Realtek
                   physical id: 4
                   bus info: usb@1:4
                   version: 2.10
                   serial: 123456
                   capabilities: usb-2.10
                   configuration: maxpower=500mA speed=480Mbit/s

RinCat, 2020-01-18:
2357:0115 is already in the whitelist, are sure you loaded the module?

Noisycall, 2020-01-19:
I tried a few more things and turns out it's a Linux headers issue. 
That's my bad.
Thank you for the module though!

closed 2020-01-19 by Noisycall