RinCat/RTL88x2BU-Linux-Driver #16: Set default log level to `_DRV_WARNING_`

ogasser, 2020-04-08:
Hi @RinCat!

First of all, thanks a lot for all the effort of maintaining this kernel module! :+1:

I see a lot of messages in my journal from the driver (581 entries in just five minutes). The majority of these are from the `rtl8822b_set_FwPwrMode_cmd` function, informing about changes in the power saving mode. Example:

```
Apr 08 18:03:30 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:30 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:30 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:30 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:32 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:32 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:32 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:32 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:32 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:32 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:34 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Apr 08 18:03:34 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:34 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:45 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:45 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:45 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:45 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:47 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Apr 08 18:03:47 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:47 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:51 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Apr 08 18:03:51 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:51 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:51 mymachine kernel: RTW: Turbo EDCA =0xa42b
Apr 08 18:03:51 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:51 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:51 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:52 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Enter 802.11 power save - WIFI-LPS_CTRL_PHYDM
Apr 08 18:03:52 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:52 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = LPS, drv ps mode = 2, rlbm = 1 , smart_ps = 2, allQueueUAPSD = 0
Apr 08 18:03:54 mymachine kernel: RTW: rtw_set_ps_mode(wlp3s0u1) Leave 802.11 power save - WIFI-TRAFFIC_BUSY
Apr 08 18:03:54 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): HW port id=0
Apr 08 18:03:54 mymachine kernel: RTW: rtl8822b_set_FwPwrMode_cmd(wlp3s0u1): fw ps mode = ACTIVE, drv ps mode = 0, rlbm = 0 , smart_ps = 0, allQueueUAPSD = 0
Apr 08 18:03:54 mymachine kernel: RTW: Turbo EDCA =0x5ea42b
```

These messages are output by `RTW_INFO()` and are sent to the journal as the default log level is set to `_DRV_INFO_ = 4`.

I know I can reduce the log level via `/proc/net/rtl88x2bu/log_level`, but I would advocate for a the less verbose default log level of `_DRV_WARNING_ = 3`, as the vast majority of users is not really interested in power level changes of their wifi module and their journal gets swamped with info level output.

Would you accept a pull request to address this?

Thanks!

RinCat, 2020-04-08:
Done, there is a debug level could be configured in Makefile.
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/9c6c2539939ff8208df5ba3cf2224fa5173ab49c

ogasser, 2020-04-08:
Thanks a lot 👍

closed 2020-04-08 by ogasser