RinCat/RTL88x2BU-Linux-Driver #19: TPLink Archer T3U Plus

leviwaiu, 2020-06-29:
Seeing that there are no linux-related discussions on the internet about this adapter at the moment I am betting based on the name in that it is similar to the T3U.

Name: TP-Link Archer T3U Plus
https://www.tp-link.com/uk/home-networking/adapter/archer-t3u-plus/#overview

lsusb:
`Bus 003 Device 003: ID 2357:0138 TP-Link 802.11ac NIC`

RinCat, 2020-06-29:
It should be 8812bu from windows driver inf.
I have updated this driver and you can have a try.

leviwaiu, 2020-06-29:
Seems to work after installation and a reboot. Thanks.

closed 2020-06-29 by leviwaiu

splitcells, 2021-11-18:
Did the [Manual installation](https://github.com/RinCat/RTL88x2BU-Linux-Driver#manual-installation) instructions not work?
What were the problems?