RinCat/RTL88x2BU-Linux-Driver #22: Which driver?

djzort, 2020-07-17:
It seems there are similar efforts on this driver at https://github.com/cilynx/rtl88x2bu
Is there any relationship between the two efforts? Which is "best"?

RinCat, 2020-07-17:
Both are based on Realtek official open source driver. The version is also same (you need change branch to 5.6.1.6 in cilynx).
While cilynx version do have some extra patches, I prefer keep the original as long as it not broken.
Both should works fine in most cases.

djzort, 2020-07-17:
Ok wow thanks for the fast response. Can you comment on this? https://github.com/EntropicEffect/rtl8822bu

RinCat, 2020-07-17:
That's 2 years old 5.3.1 version.

Polve, 2020-07-17:
Is there any possibility to push this upstream?

RinCat, 2020-07-17:
@Polve which upstream? Realtek? They only release the final source code. There is no repo outside the company people can access.

Polve, 2020-07-17:
I was thinking about the linux kernel: there are many realtek drivers, why not this one?

RinCat, 2020-07-17:
@Polve Those drivers were not suppose to put in kernel at first place, thus the code design and quality do not meet the kernel requirements. They may need rewrite a lot of code to do so. 
https://www.phoronix.com/scan.php?page=news_item&px=Realtek-RTW88-WiFi-Linux-Driver

Polve, 2020-07-17:
Thanks @RinCat  for the prompt reply!

closed 2020-07-18 by djzort