RinCat/RTL88x2BU-Linux-Driver #23: manjaro kernel 5.8rc5 not working

mirsella, 2020-07-21:
Hello

the driver works for me on the earlier version, including the version I was with (5.7) before trying 5.8.
so after installing the kernel 5.8rc5 I rebuilt the AUR package with yay on manjaro, no error that seems important.
after a reboot, the card is not recognized, as if there was no driver. no wifi options from the NetworkManager.
so is someone have their wireless adapter working with the 5.8 kernel ? especially with manjaro.
thanks

RinCat, 2020-07-21:
5.8 is not tested yet. I usually wait until it release stable.
Maybe you can post compilation log? Also check if any error in dmesg.

mirsella, 2020-07-21:
yes, after reflection it's was dumb to create a new issue with an experimental kernel. don't hesitate to close the issue if you want.

if you want to try to troubleshoot : 

tried to compile with linux58, failed with this error : https://pastebin.com/wQYmrZe5
rebooted to linux57, compiled successfully with only some warning, working on linux57 like before.
I'm a noob, but maybe if it's throw this error at compile time with linux58, it would fails too when compiled successfully but run on linux58 ?
`sudo dmesg` output on linux58 (compiled on linux57) : https://pastebin.com/pqd5yfkz
maybe you want a `journalctl -p 3 -xb -1` output too ? i have to reboot to linux58, tell me if you want another command.

PS : i don't know if it's because of the driver or the adapter, but it's really tricky with the USB port. doesn't work with some of USB3.1 gen2 directly on my motherboard and don't work with my front port on the case.
only work for classic usb3.1 of the motherboard, didn't tried with USB2.

mirsella, 2020-07-21:
it work ! hacky, but it work **for** me

after i got the error i rushed to post the output etc. but didn't even read the error.
just after i read the error. (remember I'm  a noob)

i saw the error was `sha256_state` being declared a second time, which make it exit on linux58 but not linux57, weird.
so i commented the second declaration of `sha256_state` in the `/include/rtw_security.h` file. fixed this issue, but another one came, which i think have nothing to do with the first one.

second error, here the output : https://pastebin.com/r6RD53KB
the error seems to be `.mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,` because `error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’`.
my noob fix was to comment the line in `os_dep/linux/ioctl_cfg80211.c` lol.

so after commenting those 2 problematic statement, i runned `make` again and it worked.
I'll give you the output anyway if you want it : https://pastebin.com/dkpxgAaP
it work now the linux58rc5 of manjaro.

if you can explain to me what was `.mgmt_frame_register` used for ? thanks.

i don't fix the problem i got were bug due to the kernel being experimental, but a change in the behaviour of something ? the compiler ? 

i didn't close the issue because i thought the issue is still here for other people. my issue is fixed, so feel free to close it if you want to.

thanks

RinCat, 2020-07-21:
Their are multiple kernel APIs changed, give me some time to check.

RinCat, 2020-07-21:
You can have a try the new version, I did not test it since I don't have 5.8 kernel atm.

mirsella, 2020-07-22:
same error : https://pastebin.com/iN3TmfTq
by commenting the line 9938 it compiles.
I tried compiling it from linux58, maybe if I compile it from linux57 it will work for linux58 ? i'll try

mirsella, 2020-07-22:
so i rebooted to try on linux57, but WTF.
I've `sudo make uninstall` before reboot, i rebooted and still wifi.
I've `sudo make uninstall` while running Linux58 and Linux 57, but both when i reboot there no wifi on Linux 58, but it work on Linux57. make uninstall finished without error too.

RinCat, 2020-07-22:
A small error, now should fixed.

mirsella, 2020-07-22:
will try when I got my pc

mirsella, 2020-07-23:
yep it work for me. thanks

closed 2020-07-23 by mirsella