RinCat/RTL88x2BU-Linux-Driver #27: changing txpower

jmannt3ther, 2020-08-02:
With this driver, no matter what we do the output of:

iw dev or iw wlan0 info

is: txpower 12.00 dBm

when trying to set: (after forcing down)
iw dev wlan0 set txpower fixed 20mBm or 30mBm, nothing changes.

The makefile has these options:

CONFIG_TXPWR_BY_RATE = y
CONFIG_TXPWR_BY_RATE_EN = y
CONFIG_TXPWR_LIMIT = y
CONFIG_TXPWR_LIMIT_EN = n

But they don't seem to change anything.

Has anyone been able to set the txpower to 20mBm or 30 or something other than 12?

RinCat, 2020-08-02:
txpower is not implemented in the official driver.
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/os_dep/linux/ioctl_cfg80211.c#L4180

RinCat, 2020-08-02:
Lucky for you, I just updated to the last official driver I can found, and it "seems" support change TX power.
It can be set and read, but I have no way to test if it actually changed TX power.

Use `iw wlan0 set txpower fixed 2000` to change to 20.00 dBm.

jmannt3ther, 2020-08-02:
Compiled and working. Seems it actually sets it on the adapter itself. I went back to the old driver on another OS and it was now 20dbm. Updating the driver definitely changed it from 12 to 20.  But this driver DOES let you change it. Thanks so much!

jmannt3ther, 2020-08-02:
When running hostapd we are seeing:
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: interface state COUNTRY_UPDATE->HT_SCAN

RinCat, 2020-08-03:
That seems normal, it need to check country regulations.

jmannt3ther, 2020-08-05:
Hi, so after running a test with 2 clients overnight we got all these errors hostapd stopped working. Have you seen this before?

[dmesg_log.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5029126/dmesg_log.txt)

RinCat, 2020-08-05:
No, but the code shows the error is H2C API command timeout. Is the device under heavy load?

jmannt3ther, 2020-08-05:
At that time no, but sometimes yes. How can we prevent it? Are there any run-time flags or anything we can set?

jmannt3ther, 2020-08-05:
I feel like this wasn't an issue in the older driver. Can you tell me where I can get your 5.6 version and do a comparison test?

RinCat, 2020-08-05:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/archive/314b662331971b10eb484fbcedb06bf5abfcb211.zip
Last 5.6 version.

jmannt3ther, 2020-08-05:
Any ideas on those errors?

jmannt3ther, 2020-08-05:
The manufacturer also provided me with an updated hostapd/wpa supplicant if that helps.

RinCat, 2020-08-05:
No ideas, since I have never encountered this bug. You may try unload and load the driver again to see if it reset.
hostapd/wpa should be fine.

jmannt3ther, 2020-08-05:
How about this error: RTW: ERROR rtl8822b_c2h_handler_no_io: C2H(9) enqueue FAIL!

jmannt3ther, 2020-08-05:
How can we disable btrtl (bluetooth driver for same wireless card) and btusb for that card?

RinCat, 2020-08-05:
> RTW: [HALMAC] [ALWAYS]shall R reg twice!!
> RTW: ERROR [HALMAC] [ERR] Dump efuse in suspend
is normal. But 
> RTW: ERROR rtl8822b_c2h_handler_no_io: C2H(9) enqueue FAIL!
is not.

RinCat, 2020-08-05:
Bluetooth is disabled, or you have anything in `/proc/net/rtl88x2bu/wlan0/btcoex`?

jmannt3ther, 2020-08-05:
its empty

jmannt3ther, 2020-08-05:
we can recompile with CONFIG_BT_COEXIST = n instead of yes and try it also...

RinCat, 2020-08-05:
You can give it a try

jmannt3ther, 2020-08-05:
Any issue setting this in makefile?
CONFIG_POWER_SAVING = n

We also set:
iw dev wlan0 set power_save off

RinCat, 2020-08-05:
you can turn it off

jmannt3ther, 2020-08-06:
So today we have the following failure on the latest 5.8 driver:

`RTW: WARN _beamforming_leave: ` 

and then it deauth's the wifi device...and no longer allow anything to connect without a reboot.

RinCat, 2020-08-06:
Set `rtw_beamform_cap=0` to disable beamform. But I think you probably crashed the device for some reason.
Make sure it has enough power supply and not overheat.

jmannt3ther, 2020-08-07:
still getting these errors which break the ap...? any ideas?

[75756.600303] RTW: WARN _beamforming_leave: xx:xx:xx:xx:xx:xx is neither beamforming ee or er!!
[75756.600309] RTW: WARN _beamforming_leave: xx:xx:xx:xx:xx:xx is neither beamforming ee or er!!

RinCat, 2020-08-07:
That means the mac address interface is not found.

jmannt3ther, 2020-08-07:
mac for what? why does it break the ap?

RinCat, 2020-08-08:
Maybe the interface disappeared for some reason. It cannot found an interface with the mac address.

jmannt3ther, 2020-08-11:
Hi, so our adapter is usb 2.0 and no matter what modprobe command comes up it seems to be trying usb 3.0 mode. is there any other way to force usb 2.0 other than during modprobe? such as in the makefile?

RinCat, 2020-08-11:
No, and the default is not switch unless you set `rtw_switch_usb_mode=1`.

jmannt3ther, 2020-08-13:
In another bug #29 you reference a setting in the makefile that can enable usb 3.0. can you let us know what that is? our issues are related to when running under a virtual machine (virtualbox) the usb 3.0 xhci driver has errors. We need to disable usb 3.0. Thanks!

RinCat, 2020-08-13:
There is no config in makefile, the place you can change USB version is use a [module parameter](https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/os_dep/linux/os_intfs.c#L391).
And the default is using USB 2.0 and does not upgrade to USB 3.0 unless you set the parameter.

jmannt3ther, 2020-08-13:
OK awesome, we will check. Any other ideas related to xhci drivers? such as the option for CONFIG_USB_HCI?

RinCat, 2020-08-13:
That is USB Host Controller Interface, nothing to do with USB3.0.

jmannt3ther, 2020-08-13:
We are seeing this exact issue someone else is reporting when running the driver under virtualbox on alpine, under high load only. all other times it works great. 

https://www.virtualbox.org/ticket/19426

RinCat, 2020-08-13:
It more likely a bug in the kernel, try update it.

jmannt3ther, 2020-08-13:
Running the latest alpine offers. 5.x. Still have the same issues. Trying to come up with something.....

RinCat, 2020-08-13:
Is your USB Controller ASMedia ASM1142?

jmannt3ther, 2020-08-14:
No Intel Corporation 7 Series/C210 Series Chipset Family USB, i believe based on lshw

jmannt3ther, 2020-08-17:
have you seen this issue?

RTW: ERROR rtl8822b_c2h_handler_no_io: C2H(9) enqueue FAIL!

RinCat, 2020-08-18:
That is usually IO error

jmannt3ther, 2020-08-19:
so we got our issues narrowed down. the only real issue we see is when there is heavy load we get these:

RTW:  ERROR _btmpoper_cmd: Didn't get h2C Rsp Event!

RinCat, 2020-08-19:
Well, it means command timeout, may be the hardware overheated. You need to know this chip is kind of low-end, most of those adapters don't have a good heat dissipation efficiency.

jmannt3ther, 2020-08-19:
I think that this might be caused by allowing hostapd to increase the tx power. It seems the default for the card is 12dbm. if we set a fixed value, it seems that hostapd is increasing it to 20dbm....how can we disable letting hostapd change it?

jmannt3ther, 2020-08-19:
Also we are seeing this now:
wlan0: interface state HT_SCAN->DISABLED

RinCat, 2020-08-19:
I don't think hostapd will change TX power, but you can set it after running hostapd.

interface state HT_SCAN->DISABLED is complex issues, can be cause by many things. you need more logs.

jmannt3ther, 2020-08-19:
well i set it like this and its being changed by hosatpd:

iw wlan0 set txpower fixed 1200

RinCat, 2020-08-19:
I don't think hosatpd allowed user set it manually.
Try `iw wlan0 set txpower limit 1200`?

jmannt3ther, 2020-08-19:
is there a reason that after a while everything would just break?

jmannt3ther, 2020-08-19:
also older drivers didn't set this, i think you added the ability for it to set dynamically, based on a previous request we had.

jmannt3ther, 2020-08-19:
After setting txpower it shows as this:

[image]

jmannt3ther, 2020-08-19:
can we disable the power setting in makefile?

RinCat, 2020-08-19:
No, the code don't change TX power unless something set it to the value. 
And you can't disable power setting in makefile, but you can commit it out 
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/os_dep/linux/ioctl_cfg80211.c#L4357
Line 4357 to 4386.

jmannt3ther, 2020-08-26:
So we are seeing everything work great but after a few hours it just stops, and nothing associates. I believe it to be a power save related issue.  Best way to disable in make file? Or also somewhere else?

we are seeing: 
[image]

jmannt3ther, 2020-08-26:
Such as RTW_WOW_LPS_MODE=0

RinCat, 2020-08-26:
There is a `CONFIG_POWER_SAVING` in Makefile

jmannt3ther, 2020-08-27:
Have you seen this error before?
RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL

RinCat, 2020-08-27:
No, but that means the target wifi network not exists.

jmannt3ther, 2020-08-27:
Is there a good reference for these errors?

RinCat, 2020-08-27:
The error is happened [here](https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/master/core/rtw_mlme.c#L1999). But don't have much information, only told you system can't find target wifi network.

[24 more replies]