RinCat/RTL88x2BU-Linux-Driver #29: Cant connect to USB 3.0 like on Windows 10

HeXN0P, 2020-08-04:
*-usbhost:0
                product: xHCI Host Controller
                vendor: Linux 5.8.0-1-MANJARO xhci-hcd
                physical id: 0
                bus info: usb@2
                logical name: usb2
                version: 5.08
                capabilities: usb-2.00
                configuration: driver=hub slots=4 speed=480Mbit/s
              *-usb
                   description: USB hub
                   product: USB2.0 Hub
                   vendor: VIA Labs, Inc.
                   physical id: 2
                   bus info: usb@2:2
                   version: 2.14
                   capabilities: usb-2.10
                   configuration: driver=hub slots=4 speed=480Mbit/s
                 *-usb
                      description: Generic USB device
                      product: 802.11ac NIC
                      vendor: Realtek
                      physical id: 3
                      bus info: usb@2:2.3
                      version: 2.10
                      serial: 123456
                      capabilities: usb-2.10
                      configuration: driver=rtl88x2bu maxpower=500mA speed=480Mbit/s

I connect usb wifi via a USB 3.0 HUB. On Windows 10 after connect to an AC 5Ghz network it autoswitch to USB 3.0 connection but not with linux driver ? The driver not support or some misconfiguration ?

HeXN0P, 2020-08-04:
Bus 002 Device 004: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 RTL88x2bu [AC1200 Techkey]
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x0006
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

LSUSB output

HeXN0P, 2020-08-04:
[image]
In Windows 10 same usb port same wifi station

RinCat, 2020-08-05:
More likely your usbhost and usb hub are running at USB 2.0. Try speed test to see the actually speed.

RinCat, 2020-08-05:
I did some tests, it seems not working. You can use `rtw_switch_usb_mode=1` to force it upgrade to 3.0, but it stuck in restart loop.

HeXN0P, 2020-08-05:
Bus 004 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 RTL88x2bu [AC1200 Techkey]
  bcdDevice            3.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0053
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              504mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11ac NIC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         287 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0004
  (Bus Powered)
  U1 Enabled

Thanks you after edit rtw_switch_usb_mode=1 i can using at USB 3.0 speed right now. No restart loop.

HeXN0P, 2020-08-05:
[image]

HeXN0P, 2020-08-05:
*-usbhost:1
                product: xHCI Host Controller
                vendor: Linux 5.8.0-1-MANJARO xhci-hcd
                physical id: 1
                bus info: usb@4
                logical name: usb4
                version: 5.08
                capabilities: usb-3.00
                configuration: driver=hub slots=4 speed=5000Mbit/s
              *-usb
                   description: USB hub
                   product: USB3.0 Hub
                   vendor: VIA Labs, Inc.
                   physical id: 2
                   bus info: usb@4:2
                   version: 2.14
                   capabilities: usb-3.10
                   configuration: driver=hub slots=4 speed=5000Mbit/s
                 *-usb
                      description: Generic USB device
                      product: 802.11ac NIC
                      vendor: Realtek
                      physical id: 3
                      bus info: usb@4:2.3
                      version: 3.00
                      serial: 123456
                      capabilities: usb-3.00
                      configuration: driver=rtl88x2bu maxpower=504mA speed=5000Mbit/s

RinCat, 2020-08-05:
Good to know, that one doesn't work for me. But maybe different between manufacturers.

HeXN0P, 2020-08-05:
I don't know but the one i use is no brand name and i bought for 8$.

HeXN0P, 2020-08-05:
Can you add an options in Make file to make it easier for anyone who want usb 3.0 support ?

RinCat, 2020-08-05:
USB 3.0 support is default on, I can't set the switch default to 3.0 since it will broken if you plug it in 2.0. But I can add it to README.

HeXN0P, 2020-08-05:
Thanks you for your time ^^

closed 2020-08-05 by HeXN0P