RinCat/RTL88x2BU-Linux-Driver #31: Mister FPGA Compilation Failure

SumolX, 2020-08-19:
Trying to build this driver for my AC1200M which uses this chipset for the Mister FPGA platform.  The following commands produce this failure:

```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- clean
#make -C /lib/modules/5.4.0-42-generic/build M=/home/test/Projects/RTL88x2BU-Linux-Driver clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko */*/*/*.ur-safe
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko */*/*.ur-safe
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko */*.ur-safe
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko *.ur-safe
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions *.ur-safe
```

```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- KSRC=/home/test/Projects/Linux-Kernel_MiSTer modules
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /home/test/Projects/Linux-Kernel_MiSTer M=/home/test/Projects/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/home/test/Projects/Linux-Kernel_MiSTer'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  Building modules, stage 2.
  MODPOST 0 modules
make[1]: Leaving directory '/home/test/Projects/Linux-Kernel_MiSTer'
```

Thank You in advance!!!

SumolX, 2020-08-20:
Mister FPGA WiFi Compilation instructions are located here:
https://github.com/MiSTer-devel/Main_MiSTer/wiki/MISTER-CUSTOM-WIFI-DRIVER-COMPILATION-GUIDE

RinCat, 2020-08-20:
You need set `KSRC=/home/test/Projects/Linux-Kernel_MiSTer/build`. And your kernel must be prepared.

smutking, 2021-04-25:
Did you manage to build it for Mister? I tried `make prepare modules_prepare` on kernel and still got _MODPOST 0 modules_ on stage 2.