RinCat/RTL88x2BU-Linux-Driver #32: Support for DFS?

andaag, 2020-08-29:
Hi

I'm in a very active wifi area and I'm trying to see if DFS is supported. I see a lot of references to it in the code, but I can't activate any of the radar specific channels.

```
Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (radar detection)
                        * 5280 MHz [56] (20.0 dBm) (radar detection)
                        * 5300 MHz [60] (20.0 dBm) (radar detection)
                        * 5320 MHz [64] (20.0 dBm) (radar detection)
                        * 5500 MHz [100] (26.0 dBm) (radar detection)
                        * 5520 MHz [104] (26.0 dBm) (radar detection)
                        * 5540 MHz [108] (26.0 dBm) (radar detection)
                        * 5560 MHz [112] (26.0 dBm) (radar detection)
                        * 5580 MHz [116] (26.0 dBm) (radar detection)
                        * 5600 MHz [120] (26.0 dBm) (radar detection)
                        * 5620 MHz [124] (26.0 dBm) (radar detection)
                        * 5640 MHz [128] (26.0 dBm) (radar detection)
                        * 5660 MHz [132] (26.0 dBm) (radar detection)
                        * 5680 MHz [136] (26.0 dBm) (radar detection)
                        * 5700 MHz [140] (26.0 dBm) (radar detection)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)
```

```
Channel 116 (primary) not allowed for AP mode, flags: 0x3e07979 RADAR
Could not select hw_mode and channel. (-3)
wlx7ca7b0babf84: interface state UNINITIALIZED->DISABLED
wlx7ca7b0babf84: AP-DISABLED
wlx7ca7b0babf84: Unable to setup interface
```

Is this supported by the driver and or hardware?

RinCat, 2020-08-29:
DFS should supported, have you set your country?

andaag, 2020-08-30:
```
iw reg get
global
country ES: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

I've been toying with it, and I wasn't sure if it might be an issue with the setup/order of commands. So my setup script looks like:
```
iw reg set ES
rmmod 88x2bu
modprobe 88x2bu rtw_switch_usb_mode=1 rtw_country_code=ES
nmcli con up Hotspot
```

RinCat, 2020-08-30:
Which channel you wanna use?

andaag, 2020-08-30:
I got neighboors on networks up to 112, that's why I tried 116, 90% of the 5ghz traffic in the area is in 36-64. So anything outside of that would be preferable.

RinCat, 2020-08-30:
https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_or_5.9_GHz_(802.11a/h/j/n/ac/ax)
Check the link and match what bandwidth you want to use.

andaag, 2020-08-30:
I would probably try 40/80. Do I need to set this before trying those channels?

RinCat, 2020-08-30:
you can try, the default is US I think.

andaag, 2020-08-30:
Default US? For that you mean region right? I'm in Spain and that is set. Ref the output from iw reg get above.

It looks like it's currently using 80mhz.

RinCat, 2020-08-30:
The region is default US, NetworkManager maybe will set it to your local, but I didn't test that.
If you wanna use 80MHz, you can only set it to 36/52/100/116.

andaag, 2020-08-30:
I did try 116 though, that's where I get this error:
```
Channel 116 (primary) not allowed for AP mode, flags: 0x3e07979 RADAR
```

andaag, 2020-08-30:
I also can't get 52 connected:
```
Channel 52 (primary) not allowed for AP mode, flags: 0x979 RADAR
Could not select hw_mode and channel. (-3)
wlx7ca7b0babf84: interface state UNINITIALIZED->DISABLED
wlx7ca7b0babf84: AP-DISABLED
wlx7ca7b0babf84: Unable to setup interface.
RTW: nolinked power save leave
Failed to initialize AP interface
wlx7ca7b0babf84: interface state DISABLED->DISABLED
wlx7ca7b0babf84: AP-DISABLED
hostapd_free_hapd_data: Interface wlx7ca7b0babf84 wasn't started
wlx7ca7b0babf84: CTRL-EVENT-DISCONNECTED bssid=7c:a7:b0:ba:bf:84 reason=3 locally_generated=1
```

RinCat, 2020-08-30:
check if you have installed `wireless-regdb` and `crda`

andaag, 2020-08-30:
```
# dpkg --get-selections | grep -i crda
crda                                            install
# dpkg --get-selections | grep -i wireless-reg
wireless-regdb                                  install
```

RinCat, 2020-08-30:
That's weird. 
`36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140`
are allowed in the driver for ETSI1 region, which included ES.
Which channel you can use?

andaag, 2020-08-30:
36, 40, 44, 48 = works
52 = Channel 52 (primary) not allowed for AP mode, flags: 0x979 RADAR
56 = Channel 56 (primary) not allowed for AP mode, flags: 0x1979 RADAR
60 = Channel 60 (primary) not allowed for AP mode, flags: 0x3179 RADAR
64 = Channel 64 (primary) not allowed for AP mode, flags: 0x6179 RADAR
100 = Channel 100 (primary) not allowed for AP mode, flags: 0x301979 RADAR
I can see a bit of a pattern here, so I'll skip ahead a bit. If you want me to test the rest let me know, it just takes me a while to get back to you with the details!
124 = Channel 124 (primary) not allowed for AP mode, flags: 0xf807979 RADAR
136 = Channel 136 (primary) not allowed for AP mode, flags: 0xc006179 RADAR
140 = Channel 140 (primary) not allowed for AP mode, flags: 0x8004169 RADAR

These are tested by modifying networkmanager and letting that setup the whole chain, including wpa_supplicant etc.

I can also execute this without getting any errors. I'm not sure if I'm supposed to get errors at this point, or when activating it with wpa supplicant?
```
iwconfig wlx7ca7b0babf84 channel 116
```

andaag, 2020-08-30:
I also noticed this when starting up on channel 40 with network manager:
```
RTW: ============ STA [30:fd:38:c9:a4:4e]  ===================
RTW: mac_id : 0
RTW: wireless_mode : 0x14
RTW: mimo_type : 0
RTW: static smps : N
RTW: bw_mode : 20MHz, ra_bw_mode : 20MHz
RTW: rate_id : 10
RTW: rssi : -1 (%), rssi_level : 0
RTW: is_support_sgi : Y, is_vht_enable : N
RTW: disable_ra : N, disable_pt : N
RTW: is_noisy : N
RTW: txrx_state : 0
RTW: curr_tx_rate : CCK_1M (L)
RTW: curr_tx_bw : 20MHz
RTW: curr_retry_ratio : 0
RTW: ra_mask : 0x00000000000ffff0
RTW: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:AES
```

Maybe networkmanager is forcing something to 20mhz? And that's what breaks things?

andaag, 2020-08-30:
Testing this with hostapd instead to see if it's a networkmanager issue:
```
hostapd -d hostapd.conf
random: getrandom() support available
Configuration file: hostapd.conf
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: interface wlx7ca7b0babf84 in phy phy14
nl80211: Set mode ifindex 18 iftype 3 (AP)
nl80211: Setup AP(wlx7ca7b0babf84) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x557dd5d02ce0 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=0501
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=0503
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=0504
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=12
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x557dd5d02ce0 match=7f
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x557dd5d02ce0 match=
nl80211: Enable Probe Request reporting nl_preq=0x557dd5d07720
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x557dd5d07720 match=
rfkill: initial event: idx=14 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 18 (ifidx_reason -1)
nl80211: if_indices[16]: 18(-1)
phy: phy14
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlx7ca7b0babf84: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code ES, new country code ES 
nl80211: Regulatory information - country=ES (DFS-ETSI)
nl80211: 2400-2483 @ 40 MHz 20 mBm
nl80211: 5150-5250 @ 80 MHz 23 mBm (no outdoor)
nl80211: 5250-5350 @ 80 MHz 20 mBm (no outdoor) (DFS)
nl80211: 5470-5725 @ 160 MHz 26 mBm (DFS)
nl80211: 5725-5875 @ 80 MHz 13 mBm
nl80211: 57000-66000 @ 2160 MHz 40 mBm
nl80211: Added 802.11b mode based on 802.11g information
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[DISABLED]
nl80211: Mode IEEE 802.11a: 5180 5200 5220 5240 5260[RADAR] 5280[RADAR] 5300[RADAR] 5320[RADAR] 5500[RADAR] 5520[RADAR] 5540[RADAR] 5560[RADAR] 5580[RADAR] 5600[RADAR] 5620[RADAR] 5640[RADAR] 5660[RADAR] 5680[RADAR]
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[DISABLED]
Channel 116 (primary) not allowed for AP mode, flags: 0x3e07979 RADAR
wlx7ca7b0babf84: IEEE 802.11 Configured channel (116) not found from the channel list of current mode (2) IEEE 802.11a
wlx7ca7b0babf84: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlx7ca7b0babf84: interface state COUNTRY_UPDATE->DISABLED
wlx7ca7b0babf84: AP-DISABLED 
wlx7ca7b0babf84: Unable to setup interface.
hostapd_interface_deinit_free(0x557dd5cff9a0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x557dd5cff9a0)
wlx7ca7b0babf84: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlx7ca7b0babf84
wlx7ca7b0babf84: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlx7ca7b0babf84 ff:ff:ff:ff:ff:ff --> -22 (Invalid argument)
wlx7ca7b0babf84: AP-DISABLED 
hostapd_cleanup(hapd=0x557dd5d01210 (wlx7ca7b0babf84))
wlx7ca7b0babf84: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlx7ca7b0babf84 wasn't started
hostapd_interface_deinit_free: driver=0x557dd47127e0 drv_priv=0x557dd5d02480 -> hapd_deinit
nl80211: deinit ifname=wlx7ca7b0babf84 disabled_11b_rates=0
nl80211: Disable Probe Request reporting nl_preq=0x8888ddf55d58ffa9
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=18)
netlink: Operstate: ifindex=18 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 18 iftype 2 (STATION)
nl80211: Teardown AP(wlx7ca7b0babf84) - device_ap_sme=1 use_monitor=0
nl80211: Unsubscribe mgmt frames handle 0x8888ddf55d58a469 (AP teardown (dev SME))
hostapd_interface_free(0x557dd5cff9a0)
hostapd_interface_free: free hapd 0x557dd5d01210
hostapd_cleanup_iface(0x557dd5cff9a0)
hostapd_cleanup_iface_partial(0x557dd5cff9a0)
hostapd_cleanup_iface: free iface=0x557dd5cff9a0
```

hostapd.conf:
```
interface=wlx7ca7b0babf84
# "a" simply means 5GHz
hw_mode=a
# the channel to use, 0 means the AP will search for the channel with the least interferences (ACS)
channel=116
#vht_oper_chwidth=1 #0 = 20/40, 1 = 80, 2 = 160, 3 = 80+80
ieee80211d=1
country_code=ES
ieee80211n=1
# 802.11ac support
ieee80211ac=1         
wmm_enabled=1

ssid=<XXXX>
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=<XXXX>
```

RinCat, 2020-08-30:
I did some local test, it seems hostapd not allowed set up AP in radar/DFS channel:
```
nl80211: Mode IEEE 802.11a: 5180 5200 5220 5240 5260[RADAR] 5280[RADAR] 5300[RADAR] 5320[RADAR] 5500[RADAR] 5520[RADAR] 5540[RADAR] 5560[RADAR] 5580[RADAR] 5600[RADAR] 5620[RADAR] 5640[RADAR] 5660[RADAR] 5680[RADAR]
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=23 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=23 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=23 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=23 dBm
```
The driver itself is allowed

RinCat, 2020-08-30:
You can bypass this if you 100% sure it is legal channel, set to a country that allowed use the channels.

andaag, 2020-08-30:
I'm sure, as my ISP's router (from vodafone, which is pretty major in Spain) came with these enabled. + 3 of my neighboors are using these channels with wifi names like "vodafone_xxx" and "movistar_xxx" (another major provider).

I'll look into this now that I know it's hostapd blocking it. I'm a bit uncertain about how to fix this in the best way... I don't want to change crda as I'm sure that would disable the radar check, and possibly skip dfs, which is not legal.

I'll close this and I'll follow up elsewhere. Thanks a *lot* for your help debugging!

andaag, 2020-08-30:
I'm spying a bit on the source for this:

```
/*
			 * Disable all channels that are marked not to allow
			 * to initiate radiation (a.k.a. passive scan and no
			 * IBSS).
			 * Use radar channels only if the driver supports DFS.
			 */
			if ((feature->channels[j].flag &
			     HOSTAPD_CHAN_RADAR) && dfs_enabled) {
				dfs = 1;
			} else if (((feature->channels[j].flag &
				     HOSTAPD_CHAN_RADAR) &&
				    !(iface->drv_flags &
				      WPA_DRIVER_FLAGS_DFS_OFFLOAD)) ||
				   (feature->channels[j].flag &
				    HOSTAPD_CHAN_NO_IR)) {
				feature->channels[j].flag |=
					HOSTAPD_CHAN_DISABLED;
			}

			if (feature->channels[j].flag & HOSTAPD_CHAN_DISABLED)
				continue;

			wpa_printf(MSG_MSGDUMP, "Allowed channel: mode=%d "
				   "chan=%d freq=%d MHz max_tx_power=%d dBm%s",
				   feature->mode,
				   feature->channels[j].chan,
				   feature->channels[j].freq,
				   feature->channels[j].max_tx_power,
				   dfs ? dfs_info(&feature->channels[j]) : "");
		}
```

dfs_enabled points at this:
```
int dfs_enabled = hapd->iconf->ieee80211h &&
			(iface->drv_flags & WPA_DRIVER_FLAGS_RADAR);
```

I added some debug info along with dfs_enabled:
```
hapd->iconf->ieee80211h = 1 
iface->drv_flags & WPA_DRIVER_FLAGS_RADAR = 0
```

RinCat, 2020-08-31:
It seems you either have to override crda or patch hostapd. The DFS is handled by driver not in hostapd.

andaag, 2020-08-31:
I'm confused. What I see hostapd trying to do here is to check if the driver exposes the driver flag WPA_DRIVER_FLAGS_RADAR, and if it does enable it's DFS support.

That flag should be coming from the driver right?

RinCat, 2020-08-31:
It is not only need a flag, hostapd want to control the DFS part, but Realtek driver not support that. It does have it own DFS implementation without relying on hostapd.

RinCat, 2020-08-31:
Realtek driver does comes with a patched hostapd 0.8, and they set all channels usable. Then let the driver takes the control.

andaag, 2020-08-31:
Ahh, that explains everything. Thank you!

closed 2020-08-31 by andaag

RinCat, 2020-08-31:
You can found one [here](https://support.dlink.ca/ProductInfo.aspx?m=DWA-181) , but hostapd 0.8 is very old.

t3therdev, 2022-04-12:
Hi, was there ever an update to this? We can't use DFS channels either....did it require a patch to hostapd? if so is there any detail or a working hostapd for this?

RinCat, 2022-04-12:
That hostapd does not has source code, so I had no idea what they changed.