RinCat/RTL88x2BU-Linux-Driver #35: Kernel oops 5.4.52: TP-Link AC1300

parthitce, 2020-09-14:
**Device**: TP-Link AC1300

```
> lsusb
Bus 001 Device 017: ID 2357:0138 TP-Link
```

Kernel SHA: [c57b1153a58a6263863667296b5f00933fc46a4f](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.4.y&id=c57b1153a58a6263863667296b5f00933fc46a4f) 
Kernel version 5.4.52
Hardware platform: imx6ULL SoC based custom board

After inserting the module manually, the device is detected and wlan0 interface is up and running. But the below kernel oops is printed on the console.

**Question:**
I pulled the current master and cross compiled manually with 5.4.52 tree. Is this a problem as the version in module is points to 5.8?

Console log:
```
# insmod ./88x2bu.ko                                                                                                                    [110/7975]
[  341.364627] RTW: module init start                                                                                                                         
[  341.368117] RTW: rtl88x2bu 5.4.52-00003-g212cb315f232                                                                                                      
[  341.373727] RTW: build time: Sep 14 2020 12:17:16                                                                                                          
[  341.378497] RTW: rtl88x2bu BT-Coex version = COEX20191120-7777                                                                                             
[  341.396587] RTW: [HALMAC]11692M                                                                                                                            
[  341.396587] HALMAC_MAJOR_VER = 1                                                                                                                           
[  341.396587] HALMAC_PROTOTYPE_VER = 6                                                                                                                       
[  341.396587] HALMAC_MINOR_VER = 5                                                                                                                           
[  341.396587] HALMAC_PATCH_VER = 6*                                                                                                                          
[  341.423948] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend                                                                                                  
[  342.516425] RTW: HW EFUSE                                                                                                                                  
[  342.519142] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00                                                                                 
[  342.526762] RTW: 0x010: 2D 2D 2E 2E  2E 2E 2B 2C  2C 2D 2D 14  00 00 FF FF                                                                                 
[  342.534188] RTW: 0x020: FF FF 25 24  24 24 23 23  23 23 24 26  28 28 28 28                                                                                 
[  342.541497] RTW: 0x030: 13 00 FF FF  00 FF 1C 00  FF FF 2C 2D  2D 2E 2E 2E                                                                                 
[  342.548741] RTW: 0x040: 2C 2C 2C 2D  2D 14 00 00  FF FF FF FF  29 27 26 26   
[  342.556042] RTW: 0x050: 25 25 24 24  25 26 27 28  28 28 13 00  FF FF 00 FF  
[  342.563333] RTW: 0x060: 1C 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.570617] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.577812] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.585082] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.592351] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.599533] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  A6 2F 1E 00  FF FF FF FF  
[  342.606804] RTW: 0x0C0: FF 01 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF   
[  342.614079] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.621352] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.628536] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.635815] RTW: 0x100: 57 23 38 01  E1 67 02 98  48 27 7A 47  48 09 03 52  
[  342.643082] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63   
[  342.650352] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF   
[  342.657537] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.664801] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF   
[  342.672069] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.679252] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.686518] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.693791] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.701067] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.708249] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.715520] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.722789] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[  342.730057] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.737238] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.744501] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.751771] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.758955] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.766221] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.773561] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.780858] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.788050] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[  342.795326] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF   
[  342.802608] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... 71 lines cut ...]
[  343.616828] [<c0103154>] (do_one_initcall) from [<c01d947c>] (do_init_module+0x5c/0x224)
[  343.624955] [<c01d947c>] (do_init_module) from [<c01d8388>] (load_module+0x1f4c/0x2590)
[  343.632991] [<c01d8388>] (load_module) from [<c01d8c10>] (sys_finit_module+0xa4/0xd8)
[  343.640853] [<c01d8c10>] (sys_finit_module) from [<c0101000>] (ret_fast_syscall+0x0/0x28)
[  343.649051] Exception stack(0xc8b15fa8 to 0xc8b15ff0)
[  343.654132] 5fa0:                   e450bb00 beaa4e14 00000003 01ffb190 00000000 beaa4f19
[  343.662337] 5fc0: e450bb00 beaa4e14 00000002 0000017b 00000000 00000000 0053ea84 00000000
[  343.670534] 5fe0: beaa4c70 beaa4c60 004c8f7f b6e2c942
[  343.711720] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  343.719365] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  343.725556] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  343.732804] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[  343.740659] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[  343.746802] RTW: [RF_PATH] Registry's RF PATH:UNKNOWN
[  343.752179] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[  343.758575] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R
[  343.763683] RTW: [RF_PATH] NumTotalRFPath:2
[  343.767906] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[  343.773404] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[  343.778842] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[  343.793391] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  343.813254] RTW: rtw_ndev_init(wlan0) if1 mac_addr=98:48:27:7a:47:48
[  343.839030] usbcore: registered new interface driver rtl88x2bu
[  343.845157] RTW: module init ret=0
```

matthiasklein, 2020-09-14:
I also see the deadlock on different i.MX6 platforms. I have tried several versions of the Realtek driver with kernel 4.14 and 5.7.1.

But until now I have no idea what the reason for the message is.

parthitce, 2020-09-14:
> I also see the deadlock on different i.MX6 platforms. I have tried several versions of the Realtek driver with kernel 4.14 and 5.7.1.
> 
> But until now I have no idea what the reason for the message is.

@matthiasklein Thanks for your update. I hope still this is something to do with the Realtek driver itself? If not, it makes sense to open a thread in community.nxp.com as well.

matthiasklein, 2020-09-14:
@parthitce Yes, I think it is a problem in the Realtek driver, which is either caused by " unfavorable" combinations in the kernel options or by timing effects in the IMX.6 USB host driver.

But I do not know ...

What does your hardware look like? Is it based on a commercially available SoM? Can you share your kernel configuration?

parthitce, 2020-09-14:
> What does your hardware look like? Is it based on a commercially available SoM? Can you share your kernel configuration?

@matthiasklein Thanks, yes it is based on [Phytec imx6ULL SoM with NAND](https://www.phytec.de/produkt/system-on-modules/phycore-imx-6-ul/).

Kernel version: [Mainline stable 5.4.52](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.4.y&id=c57b1153a58a6263863667296b5f00933fc46a4f) (with custom patches for custom board device tree alone, nothing much here)
Defconfig: Mainline [imx_v6_v7_defconfig](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm/configs/imx_v6_v7_defconfig?h=linux-5.4.y&id=c57b1153a58a6263863667296b5f00933fc46a4f)

RinCat, 2020-09-14:
I just tested it on 5.4.60 x86_64, and didn't see this issue. So it either ARM platform USB driver or the specific SoC USB driver had some special behavior.

parthitce, 2020-09-15:
> I just tested it on 5.4.60 x86_64, and didn't see this issue. So it either ARM platform USB driver or the specific SoC USB driver had some special behavior.

Thanks @RinCat. I will post in kernel USB mailing list and nxp community.