RinCat/RTL88x2BU-Linux-Driver #36: Driver not working in 5.8.9

sunnyhasija, 2020-09-18:
Updated to fedora 32 kernel 5.8.9, and the driver no longer works. 

Any support for the new kernel anytime soon?

ogasser, 2020-09-18:
Works for me with kernel `5.8.9-arch2-1` on Arch Linux.

sunnyhasija, 2020-09-18:
here is what I get when I try to do a `sudo make install`

```[cantos@sonnet RTL88x2BU-Linux-Driver] $ sudo make install
install -p -m 644 88x2bu.ko  /lib/modules/5.8.9-200.fc32.x86_64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.8.9-200.fc32.x86_64
[cantos@sonnet RTL88x2BU-Linux-Driver] $ modprobe 88xbu
modprobe: FATAL: Module 88xbu not found in directory /lib/modules/5.8.9-200.fc32.x86_64```

morrownr, 2020-09-18:
sunnyhasija, do I see a typo?

$ modprobe 88xbu  > $ modprobe 88x2bu

RinCat, 2020-09-18:
I have no problem use it on 5.8.10, and the module name is `88x2bu`.

sunnyhasija, 2020-09-21:
big ooof!

Sorry about the dumb issue. closing it now.

closed 2020-09-21 by sunnyhasija