RinCat/RTL88x2BU-Linux-Driver #37: v5.8.7.2.36899.20200819 Available

morrownr, 2020-09-18:
RinCat, thanks for the excellent repo. I managed to get a copy of the 5.8.7.2 source and upgraded my repo with it. My repo is: https://github.com/morrownr/88x2bu

I can upload a zip for here or you can take the changes from my repo if you are interested.

Notes:
- The new release says Realtek took support to kernel 5.6. Realtek used a 3 line solution instead of the 4 line solution most of us were using for v5.6 support. I looked it over and changed to the 3 line solution.

- I noticed one old report here is asking for monitor mode support. We'll I have been testing my repo and it seems to enter monitor mode just fine. I have yet to test further but I do have an installation of Kali and will test as I get time.

- FWIW: The reason I started a repo for the same purpose as yours is because I needed to spin up on github and back up to speed on C for a project I'm working on and this is what I came up with for training purposes. In fact, I did not notice your repo until mine was going.

Cheers

RinCat, 2020-09-18:
Thanks for report, I kind need that zip and where it from, since it should include more information than the driver.

morrownr, 2020-09-19:
[rtl88x2bu-5.8.7.2.36899.20200819.zip](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5249054/rtl88x2bu-5.8.7.2.36899.20200819.zip)

I don't seem to have recorded the source of the file. I was looking for the source for another chipset when I ran across this. I guess I download it and pressed on. A few days ago I decided to take a look. Given that there was no ReleaseNotes document from Realtek included I proceeded carefully. I used "meld" to check each line in each file that was different than my 5.8.7.1 and I came to the conclusion that this was indeed a release from Realtek. Of note: There is code to support changes in kernels 5.4 and 5.6. There are some minor additions and updates as well. There is also a bug or 2 that the community has fixed that Realtek has not fixed. My repo is building cleanly and performing very well. It isn't everyday we are working with code that Realtek released less than one month ago.

RinCat, 2020-09-20:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/2f4efae00c18aef4856725e5453060ab652c2008

Their codes still have bugs and incompatible above 5.6. Merged the new code onto the modified version.

sylphid8, 2020-12-04:
It appears that rtl88x2bu-5.8.7.4.37264.20200922 is included in the this deb https://deb.trendtechcn.com/rtl88x2bu-dkms.deb

RinCat, 2020-12-04:
@sylphid8 will have a look, but sadly no release notes available.