RinCat/RTL88x2BU-Linux-Driver #38: can fetch ssid list, but auth always failed

jezhang2014, 2020-09-20:
[WPA-PSK-CCMP+TKIP][WPA2-PSK-CCMP+TKIP][ESS]

wpa_cli:
<3>CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2447 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2447 cf2=0
<3>CTRL-EVENT-ASSOC-REJECT status_code=1
<3>CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="xxxx" auth_failures=3 duration=30 reason=CONN_FAILED
<3>CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

cf-812AC (Realtek 8812BU)
idVendor=0bda, idProduct=b812, bcdDevice= 2.10

RinCat, 2020-09-20:
I need more logs, try set `log_level DEBUG` in wpa_cli, and also dmesg logs.

jezhang2014, 2020-09-20:
Thanks! 

After I issue command: wpa_cli log_level DEBUG, it shows OK, but seems no more new log message printed out.
<3>CTRL-EVENT-SSID-REENABLED id=0 ssid="xxx"
<3>Trying to associate with ec:23:7b:35:b3:f0 (SSID='xxx' freq=2447 MHz)
<3>CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2447 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2447 cf2=0
<3>CTRL-EVENT-ASSOC-REJECT status_code=1
<3>CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="xxx" auth_failures=2 duration=20 reason=CONN_FAILED
<3>CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
<3>CTRL-EVENT-SCAN-STARTED 
<3>CTRL-EVENT-SCAN-RESULTS 
<3>CTRL-EVENT-SCAN-STARTED 
<3>CTRL-EVENT-SSID-REENABLED id=0 ssid="xxx"
<3>CTRL-EVENT-SCAN-RESULTS 
<3>Trying to associate with ec:23:7b:35:b3:f0 (SSID='xxx' freq=2447 MHz)
<3>CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2447 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2447 cf2=0
<3>CTRL-EVENT-ASSOC-REJECT status_code=1
<3>CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="xxx" auth_failures=3 duration=30 reason=CONN_FAILED
<3>CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
<3>CTRL-EVENT-SCAN-STARTED 
<3>CTRL-EVENT-SCAN-RESULTS

dmesg:
[dmsg.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5251085/dmsg.log)

RinCat, 2020-09-20:
Did you changed MAC address? dmesg log shows it cannot find adapter by MAC address.

jezhang2014, 2020-09-20:
No, I just bought this device, not change any settings, and it works well on windows and my mac...