RinCat/RTL88x2BU-Linux-Driver #42: dkms install option

nickpitlosh, 2020-09-25:
is there an option to build into dkms automatically?

RinCat, 2020-09-25:
There is a `dkms.conf`, but if you just want to try different configuration, there is no option for that.

mauromol, 2020-10-10:
Could you provide a script for non-expert users like me or at least a list of commands I have to run to install the driver as a DKMS module?
I would like to avoid to rebuild the kernel "by hand" on each kernel upgrade.
Thank you.

splitcells, 2020-10-12:
Maybe this can help you as a starting point:
* https://github.com/morrownr/88x2bu#installation-of-the-driver
* https://github.com/morrownr/88x2bu/blob/master/dkms-install.sh

mauromol, 2020-10-12:
> Maybe this can help you as a starting point:
> 
>     * https://github.com/morrownr/88x2bu#installation-of-the-driver
> 
>     * https://github.com/morrownr/88x2bu/blob/master/dkms-install.sh

Thanks for the hint. Do you know if there's any concrete difference with the driver in this repo? I'm counting at least three git repo for drivers for this WiFi adapter. Do you know why Realtek hasn't integrated it into the Linux kernel yet? By the way, isn't https://github.com/torvalds/linux/tree/master/drivers/net/wireless/realtek/rtl8xxxu supposed to support these adapters as well?

RinCat, 2020-10-13:
DKMS docs added on README.

olberger, 2023-08-06:
I guess this issue should then be closed.