RinCat/RTL88x2BU-Linux-Driver #44: is there an option to disable useless led light on a dongle?

nickpitlosh, 2020-09-25:
just thought it would be nice.

RinCat, 2020-09-25:
No built-in option, if you want to disable, try remove `#define CONFIG_RTW_LED` in `include/autoconf.h`.

gusennan, 2022-02-01:
@nickpitlosh Did you try doing this and did it work?

barak, 2022-05-04:
I'd advocate making a proper configuration option to enable/disable the LED, so it can be toggled not only at compile time but also at runtime.
```
$ echo 0 > /proc/net/rtl88x2bu/das_blinken_lights
```
This would be really useful for me, since my dongle has a super-bright blue LED and it's on at night in a little server in my bedroom and it felt like trying to sleep in a laser disco until I covered the annoying thing with some duct tape.

barak, 2022-05-04:
WAIT A SECOND!!! What's this?
```
$ cat /proc/net/rtl88x2bu/wlx1cbfceb2abdb/led_config 
strategy:12
bRegUseLed:1
iface_en_mask:0x01
ctl_en_mask[0]:0xFFFFFFFF
```