RinCat/RTL88x2BU-Linux-Driver #45: [NO ISSUE] Thanks!

uoccou, 2020-10-03:
Thank you - your RTL8x22BU driver has worked for me after a day of attempting to use others. 
Kubuntu, 5.4.0-48-generic.

Sascha324, 2020-10-08:
Works great for me too. Thank you!!
(I hope this is the right place for this)

I use Linux Mint 20 
(Linux 5.4.0-48-generic x86_64)

PVasileff, 2020-10-08:
Latest RTL8x22BU driver works great on Fedora 32 with kernel 5.8.13 using Archer T4U v3.
Thanks!

splitcells, 2020-10-10:
It worked for me too on Fedora 32, but I had to delete all stored connections and enter them again.
Maybe this is an Fedora specific thing

julianYaman, 2020-11-25:
Installed it from the [AUR](https://aur.archlinux.org/packages/rtl88x2bu-dkms-git/) and it is working with my TP-Link Archer T3U Plus AC1300 on Arch Linux, running on `5.9.10-arch1-1`. Thanks for sharing the driver!