RinCat/RTL88x2BU-Linux-Driver #46: Cant get Archer T3U to work.

garynl, 2020-10-08:
Hi,
    I have TPLink Archer T3U. It has worked in the past on older kernels, but I cant get it to run on my current system
Linux Mint Tara 19.2 MATE

```
gary@gary-laptop:~$ uname -r
5.4.0-48-generic
```
```
gary@gary-laptop:~$ lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f2:b2da Chicony Electronics Co., Ltd thinkpad t430s camera
Bus 001 Device 004: ID 0a5c:21e6 Broadcom Corp. BCM20702 Bluetooth 4.0 [ThinkPad]
Bus 001 Device 006: ID 2357:012d  
Bus 001 Device 003: ID 17ef:1003 Lenovo Integrated Smart Card Reader

```
(Its the unnamed device)
 Used git clone to get the sources, make compiles without errors or warnings, as does sudo make install

However
`sudo modprobe 88x2bu rtw_drv_log_level=5
`
just hangs. ( `sudo modprobe 88x2bu ` is exactly the same)
Cant  ctrl-C out of it and when I  reboot the machine hangs and I have to hold the on/off switch to force it to shutdown.

If I try sudo rmmod 88x2bu before I shutdown I just get a message saying the module is in use
I cant see the device using iwconfig or ifconfig

Any help would be appreciated
Gary

RinCat, 2020-10-08:
Hi, try run `dmesg -w` in a terminal, and run `modprobe 88x2bu` in another terminal, and see if any error exists.

garynl, 2020-10-08:
Well, this is embarrassing.
I cleaned my machine of all previous attempts, did an update, upgrade, git clone, make, make install, dmesg -w in one window and modprobe in another and it worked!

I can only conclude that the problem was caused by bad housekeeping on my part and the need to provide you with consistent data made me do it 'properly'.

Thanks!

closed 2020-10-08 by garynl