RinCat/RTL88x2BU-Linux-Driver #49: iw reg set XX: different results for same country

jbd1986, 2020-10-23:
Hi,

I'm using this driver to test a wireless dongle with the 8822 chipset.  The driver seems to work well in most regards, except for setting up regions. 

I’m seeing some strange behavior when switching back and forth between regions with 'iw reg set'.  It seems like not all of the channel settings go back to where they were/should be.  I was able to duplicate the results on an Nvidia Tegra TK1 system running linux 3.10, as well as a Xilinx Zynq system running 4.14.  'iw reg get' returns the expected results in each case.

For example the default country (00) has:
# iw list | grep MHz
…
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

Then if I set to (FR):
# iw reg set FR
# iw list | grep MHz
…
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

Then back to (00):
# iw reg set 00
# iw list | grep MHz
…
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

Do you see a similar situation?  Any ideas?

RinCat, 2020-10-23:
Hi, I got different results:

`iw list | grep MHz`
> * 5720 MHz [144] (disabled)
> * 5745 MHz [149] (30.0 dBm)
> * 5765 MHz [153] (30.0 dBm)
> * 5785 MHz [157] (30.0 dBm)
> * 5805 MHz [161] (30.0 dBm)
> * 5825 MHz [165] (30.0 dBm)
> * 5845 MHz [169] (disabled)
> * 5865 MHz [173] (disabled)
> * 5885 MHz [177] (disabled)

```
iw reg set FR
iw list | grep MHz
```
> * 5720 MHz [144] (disabled)
> * 5745 MHz [149] (13.0 dBm)
> * 5765 MHz [153] (13.0 dBm)
> * 5785 MHz [157] (13.0 dBm)
> * 5805 MHz [161] (13.0 dBm)
> * 5825 MHz [165] (13.0 dBm)
> * 5845 MHz [169] (disabled)
> * 5865 MHz [173] (disabled)
> * 5885 MHz [177] (disabled)

```
iw reg set 00
iw list | grep MHz
```
> * 5720 MHz [144] (disabled)
> * 5745 MHz [149] (20.0 dBm)
> * 5765 MHz [153] (20.0 dBm)
> * 5785 MHz [157] (20.0 dBm)
> * 5805 MHz [161] (20.0 dBm)
> * 5825 MHz [165] (20.0 dBm)
> * 5845 MHz [169] (disabled)
> * 5865 MHz [173] (disabled)
> * 5885 MHz [177] (disabled)

It's possible the first one is default values and not the `iw reg set` one.

jbd1986, 2020-10-23:
Are you compiling with this flag: CONFIG_RTW_IOCTL_SET_COUNTRY

You should be seeing channels get disabled and enabled.

ID is a particularly restrictive country.  What are you results for this:

iw reg set ID
# iw list | grep MHz
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (disabled)
                        short GI (80 MHz)
                        * 5180 MHz [36] (disabled)
                        * 5200 MHz [40] (disabled)
                        * 5220 MHz [44] (disabled)
                        * 5240 MHz [48] (disabled)
                        * 5260 MHz [52] (disabled)
                        * 5280 MHz [56] (disabled)
                        * 5300 MHz [60] (disabled)
                        * 5320 MHz [64] (disabled)
                        * 5500 MHz [100] (disabled)
                        * 5520 MHz [104] (disabled)
                        * 5540 MHz [108] (disabled)
                        * 5560 MHz [112] (disabled)
                        * 5580 MHz [116] (disabled)
                        * 5600 MHz [120] (disabled)
                        * 5620 MHz [124] (disabled)
                        * 5640 MHz [128] (disabled)
                        * 5660 MHz [132] (disabled)
                        * 5680 MHz [136] (disabled)
                        * 5700 MHz [140] (disabled)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (23.0 dBm)
                        * 5765 MHz [153] (23.0 dBm)
                        * 5785 MHz [157] (23.0 dBm)
                        * 5805 MHz [161] (23.0 dBm)
                        * 5825 MHz [165] (disabled)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                        * 5885 MHz [177] (disabled)

RinCat, 2020-10-23:
I just noticed you are on ARM, so I cannot test that. But you can set the `CONFIG_RTW_LOG_LEVEL` to 5 and see if something wrong.

jbd1986, 2020-10-23:
Regardless of being on ARM or x86, can you post your results for:
iw reg set ID
iw list | grep MHz

If channels are not being disabled, you will still need the CONFIG_RTW_IOCTL_SET_COUNTRY flag to be set during x86 compilation.  Otherwise channels will not be disabled on region switches.

jbd1986, 2020-10-23:
I think I may see part of the problem though.  I added some additional debug to "rtw_set_country_cmd".  

'00' is valid for wireless-regdb/crda, but appears to NOT be valid for this realtek driver:

# iw reg set 00
[   67.386258] cfg80211: Calling CRDA to update world regulatory domain
# [   67.415758] rtw_set_country_cmd input country_code is not alpha2

RinCat, 2020-10-23:
Yes, there is no "00" in the driver built-in country channel data. Which it usually let wireless-regdb/crda to control which channel can be used.

jbd1986, 2020-10-23:
I think '00' should probably be recognized as a valid region, since it's in the wireless-regdb database.  The default setup for 00 in this driver is probably incorrect if it's getting set by CRDA, but then ultimately ignored because the driver thinks it's invalid.

With that said, at least now I understand the strange behavior I was seeing.  This also means you cannot create customized regulatory domains, without patching the driver to add them.

RinCat, 2020-10-24:
I Added this support, but notice it is not use the CRDA/wireless-db data. https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/3a6f393adda93ab0bc8998c8efc9cb8e438e04b4

jbd1986, 2020-10-24:
Are you saying that the rtl88x2bu driver would then override the power/no-ir/indoor/outdoor/etc settings in db.txt in wireless-regdb?  I'll have to try modifying the db.txt on monday to see if that's the case.  I could try something simple like changing the power settings for a supported country (i.e. US).

I would expect the settings in CRDA/wireless-db to further restrict anything set by the driver.

RinCat, 2020-10-24:
They all followed the realtek built-in data set as hard limit. WiFi control system can set their limit too, but it may or may not works.

jbd1986, 2020-10-26:
It's unclear exactly what realtek is doing, but it definitely uses some of the wireless-regdb and then ALSO ignores some of it.  

I'm removing the CONFIG_RTW_IOCTL_SET_COUNTRY to test again.  NO-IR serves the same purpose as disabling channels for my application, so if that works fine, I will run without CONFIG_RTW_IOCTL_SET_COUNTRY.

jbd1986, 2020-10-26:
Even if I remove CONFIG_RTW_IOCTL_SET_COUNTRY, the realtek driver still does not completely comply with the settings in db.txt.  It no longer disables channels as you switch between countries however, which is worse.  It leaves channels which should be disabled/NO-IR in a usable condition (i.e. indonesia 'ID').

jbd1986, 2020-10-26:
I was using an older version of the driver, it seems some updates have gone in to the region settings since the version I was using (probably 12+ months ago).  Now, crda/regdb power settings seem to apply reasonably well on top of the driver settings.  Setting NO-IR in db.txt doesn't override the driver, but I could just change those settings in the driver as needed.

jbd1986, 2020-10-29:
> I Added this support, but notice it is not use the CRDA/wireless-db data. [3a6f393](https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/3a6f393adda93ab0bc8998c8efc9cb8e438e04b4)

I'm testing with this change, and it looks good, though I'm confused by the initial power limits being set for the default region '00'. They seem to be all 30dBm. I don't think that matches the internal TX power setting or crda/regdb settings. Changing to 'US' then back to '00' will give the expected crda/regdb TX power settings for '00' (20dBM / 23dBm).

RinCat, 2020-10-29:
I only found where it init, but no idea where the 30 come from. https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/3a6f393adda93ab0bc8998c8efc9cb8e438e04b4/core/rtw_mlme_ext.c#L171

jbd1986, 2020-10-30:
> I only found where it init, but no idea where the 30 come from.
> 
> https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/3a6f393adda93ab0bc8998c8efc9cb8e438e04b4/core/rtw_mlme_ext.c#L171

This code doesn't seem to run at boot.

This code does run when I use 'iw reg set' after logging in, and even in that case, I see the message:
"there is no any txpwr_regd"

jbd1986, 2020-11-30:
@RinCat FYI, I have received some hardware-region-locked dongles.  For those dongles, the enabled channels are correct when plugged in, but then get incorrectly adjusted with 'iw reg set' later on (you end up being able to enable channels that are physically disabled by the HW).  In this case, you would NOT want CONFIG_RTW_IOCTL_SET_COUNTRY flag to be set, if you want the results of 'iw list | grep MHz' to match with what the dongle is actually capable of supporting.

This would be up to the user.  I'm not sure that setting the default to Y is a good or bad idea.  Just wanted to bring this to your attention.  Basically this setting should match whether or not this dongle is region locked.

RinCat, 2020-12-03:
The 30dbm seems come from firmware
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/f9085c8a656f48022e7985f19e7113357597e952/hal/hal_com_phycfg.c#L4085