RinCat/RTL88x2BU-Linux-Driver #53: `set_fs()` will be removed in Linux 5.10

ogasser, 2020-11-06:
As [noted by Linus](http://lkml.iu.edu/hypermail/linux/kernel/2010.3/00552.html), the `set_fs()` function will be removed in Linux 5.10. There are several instances in this module where this function is used, namely in:

* os_dep/osdep_service.c
* os_dep/linux/os_intfs.c
* core/rtw_btcoex.c
* core/rtw_wlan_util.c

RinCat, 2020-11-06:
Hi, thanks for the report, but I don't have time to fix it yet, maybe this weekend will do.

ogasser, 2020-11-06:
Sure, no problem :+1:  I just wanted to give you a heads-up. Linux 5.10 is currently in the RC-2 stage and there are a couple of weeks left before it will actually be relased.

RinCat, 2020-11-20:
You can have a try. https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/64d833c13d51e0814a25a21c6fd6bedb23a15b8b

ogasser, 2020-11-24:
Latest master branch works for me on Linux 5.10rc5 :+1:

closed 2020-11-24 by ogasser