RinCat/RTL88x2BU-Linux-Driver #54: Unable to Make file

arjun02phy36, 2020-11-24:
Hi Team,

I am getting below error while trying to do make.

In file included from /root/linux_wifi/RTL88x2BU-Linux-Driver/include/drv_types.h:97,
                 from /root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c:17:
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c: In function ‘rtw_hal_set_ap_rsvdpage_loc_cmd’:
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c:6165:31: error: ‘REG_BCNQ_BDNY’ undeclared (first use in this function); did you mean ‘REG_BCNQ_BDNY_V1’?
 6165 |  header = rtw_read8(padapter, REG_BCNQ_BDNY);
      |                               ^~~~~~~~~~~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/include/rtw_io.h:418:57: note: in definition of macro ‘rtw_read8’
  418 | #define rtw_read8(adapter, addr) _rtw_read8((adapter), (addr))
      |                                                         ^~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c:6165:31: note: each undeclared identifier is reported only once for each function it appears in
 6165 |  header = rtw_read8(padapter, REG_BCNQ_BDNY);
      |                               ^~~~~~~~~~~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/include/rtw_io.h:418:57: note: in definition of macro ‘rtw_read8’
  418 | #define rtw_read8(adapter, addr) _rtw_read8((adapter), (addr))
      |                                                         ^~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c: In function ‘rtw_hal_ap_wow_enable’:
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c:6258:23: error: ‘REG_MCUTST_WOWLAN’ undeclared (first use in this function); did you mean ‘REG_MCUTST_I’?
 6258 |  rtw_write8(padapter, REG_MCUTST_WOWLAN, 0);
      |                       ^~~~~~~~~~~~~~~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/include/rtw_io.h:425:65: note: in definition of macro ‘rtw_write8’
  425 | #define  rtw_write8(adapter, addr, val) _rtw_write8((adapter), (addr), (val))
      |                                                                 ^~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c: In function ‘rtw_hal_ap_wow_disable’:
/root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.c:6283:51: error: ‘REG_MCUTST_WOWLAN’ undeclared (first use in this function); did you mean ‘REG_MCUTST_I’?
 6283 |  pwrctl->wowlan_wake_reason = rtw_read8(padapter, REG_MCUTST_WOWLAN);
      |                                                   ^~~~~~~~~~~~~~~~~
/root/linux_wifi/RTL88x2BU-Linux-Driver/include/rtw_io.h:418:57: note: in definition of macro ‘rtw_read8’
  418 | #define rtw_read8(adapter, addr) _rtw_read8((adapter), (addr))
      |                                                         ^~~~
make[2]: *** [scripts/Makefile.build:275: /root/linux_wifi/RTL88x2BU-Linux-Driver/hal/hal_com.o] Error 1
make[1]: *** [Makefile:1757: /root/linux_wifi/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-54-generic'
make: *** [Makefile:2378: modules] Error 2

Am I doing anything wrong. 
Please clarify.

Thanks,

RinCat, 2020-11-24:
Hi, what's your platform / system / compiler? 
It looks like failed include `hal_com_reg.h` which include from `hal_com.h` which include from `drv_types.h`.

arjun02phy36, 2020-11-25:
I am running on a Ubuntu 20.04 Server. I am trying to enable WOWLAN
permanently in the driver so that i can turn on the system using wifi.

Thanks,
Prasanna.

RinCat, 2020-11-25:
So what did you changed? The default is WOWLAN off.

arjun02phy36, 2020-11-25:
I just changed the below 2 attributes in MakeFile to y and tried to
compile. It was not compiling and throwing above errors. With these 2
attributes n, I am able to compile without errors.

CONFIG_WOWLAN = n
CONFIG_AP_WOWLAN = n

file: /usr/src/rtl88x2bu-5.8.7.4.37264.20200922/Makefile

arjun02phy36, 2020-11-25:
Can you please guide me on how to enable WOWLAN by default on the driver.

Thanks,

RinCat, 2020-11-25:
You probably can set `CONFIG_WOWLAN = y`, but I don't think it support `CONFIG_AP_WOWLAN`.
`CONFIG_AP_WOWLAN` can only be enable when disable `RTW_HALMAC`, but `RTW_HALMAC` is needed to use this driver.

bagustris, 2021-04-06:
Facing this problem on Centos 7.9, both gcc 4.8.5 and 7.8.1. Is there any workaround?