RinCat/RTL88x2BU-Linux-Driver #56: TP-Link Archer T4U (v3) not being recognized when plugged from an USB 3.0 extension cable

ranplayer, 2020-11-26:
Hi,

I've installed the driver via the [AUR package](https://aur.archlinux.org/packages/rtl88x2bu-dkms-git) and my Archer T4U is working fine when plugged directly through the motherboard USB ports (2.0 and 3.0). But when I plug it through a short USB 3.0 extension cable (1.8 mts) connected to any of the motherboard ports, it doesn't get recognized.

Am I missing any additional required package or configuration ? I've tried LAO executing this command `modprobe 88x2bu rtw_switch_usb_mode=1`, but it seems not to work as well.

My specs:
O.s: Manjaro Linux
Kernel: 5.4

Thanks for the good work on this project.

RinCat, 2020-11-26:
It may be that your cable is incompatible or the resistance is too large, resulting in insufficient power supply.

ranplayer, 2020-11-27:
I'm going to test some different devices as well. The cable works with my mouse and keyboard bluetooth dongles. Thanks for the quick response.