RinCat/RTL88x2BU-Linux-Driver #57: Occasionally experiencing slow upload speed

julianYaman, 2020-11-30:
Hello, 

sometimes I'm experiencing slow upload speed while the download speed remains the same.
Currently I'm using the TP-Link Archer T3U AC1300 WiFi USB adapter.

My installed OS is Arch Linux, running on kernel version `5.9.10-arch1-1`.

After closing and reopen the WiFi connection and after I replug the adapter, I have normal uplaod and download speed.

I initially thought that switching the Kernel might help but maybe I'm missing something else.

RinCat, 2020-11-30:
You should check the link speed first, as they are supposed to decrease speed if nearby devices are using the network or there are some interference. It usually has nothing to do for the kernel and driver, unless you see some error in `dmesg`.

julianYaman, 2020-11-30:
So some speedtests show the normal 100 Mbit/s down and around 35 Mbit/s up link connection speed. After some minutes, only the upload speed will drop while other devices remain having the same speed.

If there were any errors in `dmesg`, for what should I look for?

RinCat, 2020-12-01:
It's hard to say what to look for. Anything shows signal / channel / rx / tx / power etc. may give some hints.

julianYaman, 2020-12-01:
Ok, so I just `grep` the `dmesg` content and these are my results, maybe there is something relevant:

`dmesg | grep signal` showed no result

`dmesg | grep channel`
```
[    5.378131] RTW: hal_com_config_channel_plan chplan:0x26
[11591.281101] RTW: hal_com_config_channel_plan chplan:0x26
```

`dmesg | grep rx`
```
[    5.752802] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[    5.752803] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[    5.752805] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[    5.752806] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[    5.752806] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[    7.318661] RTW: txpath=0x3, rxpath=0x3
[   23.346958] RTW: txpath=0x3, rxpath=0x3
[   28.439184] RTW: txrx_state : 0
[11591.655988] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[11591.655989] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[11591.655991] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[11591.655991] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[11591.655992] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[11593.243472] RTW: txpath=0x3, rxpath=0x3
[11609.306861] RTW: txpath=0x3, rxpath=0x3
[11614.368957] RTW: txrx_state : 0

```

`dmesg | grep tx`
```
[    2.542047] r8169 0000:27:00.0 eth0: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[    5.752801] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[    5.752802] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[    5.752803] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[    5.752805] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[    5.752806] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[    5.752806] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[    5.753313] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[    7.318661] RTW: txpath=0x3, rxpath=0x3
[    7.318662] RTW: txpath_1ss:0x1, num:1
[    7.318663] RTW: txpath_2ss:0x3, num:2
[   23.346958] RTW: txpath=0x3, rxpath=0x3
[   23.346959] RTW: txpath_1ss:0x1, num:1
[   23.346960] RTW: txpath_2ss:0x3, num:2
[   28.439184] RTW: txrx_state : 0
[   28.439185] RTW: curr_tx_rate : CCK_1M (L)
[   28.439185] RTW: curr_tx_bw : 20MHz
[11591.655987] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[11591.655988] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[11591.655989] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[11591.655991] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[11591.655991] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[11591.655992] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[11591.656501] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[11593.243472] RTW: txpath=0x3, rxpath=0x3
[11593.243474] RTW: txpath_1ss:0x1, num:1
[11593.243475] RTW: txpath_2ss:0x3, num:2
[11609.306861] RTW: txpath=0x3, rxpath=0x3
[11609.306864] RTW: txpath_1ss:0x1, num:1
[11609.306864] RTW: txpath_2ss:0x3, num:2
[11614.368957] RTW: txrx_state : 0
[11614.368957] RTW: curr_tx_rate : CCK_1M (L)
[11614.368957] RTW: curr_tx_bw : 20MHz
```

`dmesg | grep power`
```
[    0.558331] thermal_sys: Registered thermal governor 'power_allocator'
[    0.616176] pci 0000:2d:00.1: D0 power state depends on 0000:2d:00.0
[    0.616190] pci 0000:2d:00.2: D0 power state depends on 0000:2d:00.0
[    0.616266] pci 0000:2d:00.3: D0 power state depends on 0000:2d:00.0
[    1.761762] usb: port power management may be unreliable
[    5.753313] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[    7.388454] audit: type=1130 audit(1606804958.163:50): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=upower comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   21.732470] RTW: nolinked power save enter
[   23.504959] RTW: nolinked power save leave
[11591.656501] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[11607.695750] RTW: nolinked power save enter
[11609.464867] RTW: nolinked power save leave

```