RinCat/RTL88x2BU-Linux-Driver #58: Kernel 5.10 RC6 Error! Bad return status for module build on kernel: 5.10.0-051000rc6-generic (x86_64)

TheJags, 2020-12-01:
Building module fails while trying to install Kernel 5.10 RC6 in Ubuntu MATE Groovy 20.10 with the following error message:

````
root@um:/usr/src/rtl88x2bu-git# sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf

root@um:/usr/src/rtl88x2bu-git# dkms add -m rtl88x2bu -v git

Creating symlink /var/lib/dkms/rtl88x2bu/git/source ->
                 /usr/src/rtl88x2bu-git

DKMS: add completed.

root@um:/usr/src/rtl88x2bu-git# dkms autoinstall

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:

cleaning build area...
'make' -j12 KVER=5.10.0-051000rc6-generic KSRC=/lib/modules/5.10.0-051000rc6-generic/build..............(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-051000rc6-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/git/build/make.log for more information.

root@um:/usr/src/rtl88x2bu-git#
````

/var/lib/dkms/rtl88x2bu/git/build/make.log

````
DKMS make.log for rtl88x2bu-git for kernel 5.10.0-051000rc6-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-051000rc6-generic/build M=/var/lib/dkms/rtl88x2bu/git/build  modules

make[1]: Entering directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.o
[... 115 lines cut ...]
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/btc/halbtc8822bwifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/btc/halbtc8822b1ant.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/btc/halbtc8822b2ant.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl88x2bu/git/build/88x2bu.o
  MODPOST /var/lib/dkms/rtl88x2bu/git/build/Module.symvers

make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/rtl88x2bu/git/build/88x2bu.ko'.  Stop.

make[3]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/88x2bu.mod.o

make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2

make[1]: *** [Makefile:1701: modules] Error 2

make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'

make: *** [Makefile:2378: modules] Error 2
````

Make and model of the wireless adapter: TP-Link Archer T3U dual-band USB wireless adapter

Many thanks.

RinCat, 2020-12-01:
Hi, I just tested and didn't see this bug.
You should check https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131

TheJags, 2020-12-01:
> Hi, I just tested and didn't see this bug.
> You should check https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131

@RinCat Many thanks for quick reply. I've submitted error and make.log on the Launchpad bug link you've provided.

Also, what would be the proper way to completely uninstall drivers/modules from your repo. Installation steps are there on the main/code page but not for the un-installation.

RinCat, 2020-12-01:
`dkms remove -m rtl88x2bu -v git --all`

TheJags, 2020-12-01:
> `dkms remove -m rtl88x2bu -v git --all`

Thanks again. Now should I close the issue or leave it open for now?

RinCat, 2020-12-01:
leave it open for now until it fixed upstream

TheJags, 2020-12-03:
Though not a solution from Ubuntu / Debian, there is a "quick fix" for the issue.

1, Add the missing file, as detailed on another GitHub repo here:

https://github.com/Mange/rtl8192eu-linux-driver/issues/205#issuecomment-736366347

and here:

https://github.com/Mange/rtl8192eu-linux-driver/issues/205#issuecomment-737891689

````
wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/your_kernel_src_folder/scripts/module.lds
sudo sed -i '$ d' /usr/src/your_kernel_src_folder/scripts/module.lds
````
Ref: https://github.com/CGarces/rtl8192eu-linux-driver/commit/30370c5ae151a36b837a80573db0cb4ba63a6d24

2, Then install the driver as detailed on the main page of this repo. I did the manual DKMS installation.

Now the driver is working just fine with kernel 5.10 RC6.

TheJags, 2020-12-14:
As described on this Launchpad thread, a fix has been released by Ubuntu:

````
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131
````
and the wireless module got built just fine for kernel 5.10, hence closing this issue.

closed 2020-12-14 by TheJags