RinCat/RTL88x2BU-Linux-Driver #59: wpa_supplicant works but not NetworkManager

jsermer, 2020-12-02:
First of all, thanks for making this arch package.  I'm not sure if I'm doing something wrong or it's some odd interaction between the adapter (https://www.amazon.com/gp/product/B07FCN6WGX/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) and NetworkManager itself.  

Using Gnome's control center to setup/configure/connect to the wireless network never connects, but using dhcpcd and the wpa_supplicant hook from the command line works (the below command will trigger dhcpcd to start wpa_supplicant and then request an IP address (can be found in arch's wiki).

```
rfkill unblock 1
sudo dhcpcd wlp5s0u2
```

My setup is as follows:

linux 5.9.11.arch2-1
rtl88x2bu-dkms-git 5.8.7.1.r107.f9085c8-1

/etc/modprobe.d/88x2bu.conf
```
options cfg80211 ieee80211_regdom=US
options 88x2bu rtw_power_mgnt=0 rtw_country_code=US rtw_switch_usb_mode=1
```

/etc/NetworkManager/conf.d/88x2bu.conf
```
[device-rtl88x2bu]
match-device=driver:rtl88x2bu
wifi.scan-rand-mac-address=no
```

/etc/wpa_supplicant/wpa_supplicant-wlp5s0u2.conf
```
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=wheel
update_config=1
country=US
network={
    bssid=myssid_bssid_mac_address
    ssid="myssid"
    psk="mypsk"
}
```

nmcli -t connection show myssid
```
connection.id:myssid
connection.uuid:39a38116-0be8-48a7-9ffc-8a378c9330c5
connection.stable-id:
connection.type:802-11-wireless
connection.interface-name:wlp5s0u2
connection.autoconnect:yes
connection.autoconnect-priority:0
connection.autoconnect-retries:-1
connection.multi-connect:0
connection.auth-retries:-1
connection.timestamp:0
connection.read-only:no
connection.permissions:
connection.zone:
connection.master:
connection.slave-type:
connection.autoconnect-slaves:-1
connection.secondaries:
connection.gateway-ping-timeout:0
connection.metered:unknown
connection.lldp:default
connection.mdns:-1
connection.llmnr:-1
connection.wait-device-timeout:-1
802-11-wireless.ssid:myssid
802-11-wireless.mode:infrastructure
802-11-wireless.band:
802-11-wireless.channel:0
[... 162 lines cut ...]
Dec 01 21:26:52 ignyter NetworkManager[659]: <info>  [1606879612.9010] Config: added 'bssid' value 'myssid_bssid_mac_address'
Dec 01 21:26:52 ignyter NetworkManager[659]: <info>  [1606879612.9011] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Dec 01 21:26:52 ignyter NetworkManager[659]: <info>  [1606879612.9011] Config: added 'psk' value '<hidden>'
Dec 01 21:26:56 ignyter NetworkManager[659]: <info>  [1606879616.1425] device (wlp5s0u2): supplicant interface state: disconnected -> scanning
Dec 01 21:26:56 ignyter NetworkManager[659]: <info>  [1606879616.1433] device (p2p-dev-wlp5s0u2): supplicant management interface state: disconnected -> scanning
Dec 01 21:27:02 ignyter NetworkManager[659]: <info>  [1606879622.3858] device (wlp5s0u2): supplicant interface state: scanning -> associating
Dec 01 21:27:02 ignyter NetworkManager[659]: <info>  [1606879622.3858] device (p2p-dev-wlp5s0u2): supplicant management interface state: scanning -> associating
Dec 01 21:27:03 ignyter NetworkManager[659]: <info>  [1606879623.8675] device (wlp5s0u2): supplicant interface state: associating -> disconnected
Dec 01 21:27:03 ignyter NetworkManager[659]: <info>  [1606879623.8676] device (p2p-dev-wlp5s0u2): supplicant management interface state: associating -> disconnected
Dec 01 21:27:17 ignyter NetworkManager[659]: <info>  [1606879637.1085] device (wlp5s0u2): supplicant interface state: disconnected -> scanning
Dec 01 21:27:17 ignyter NetworkManager[659]: <info>  [1606879637.1086] device (p2p-dev-wlp5s0u2): supplicant management interface state: disconnected -> scanning
Dec 01 21:27:18 ignyter NetworkManager[659]: <info>  [1606879638.3332] device (wlp5s0u2): supplicant interface state: scanning -> associating
Dec 01 21:27:18 ignyter NetworkManager[659]: <info>  [1606879638.3332] device (p2p-dev-wlp5s0u2): supplicant management interface state: scanning -> associating
Dec 01 21:27:18 ignyter NetworkManager[659]: <warn>  [1606879638.4727] device (wlp5s0u2): Activation: (wifi) association took too long
Dec 01 21:27:18 ignyter NetworkManager[659]: <info>  [1606879638.4728] device (wlp5s0u2): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Dec 01 21:27:18 ignyter NetworkManager[659]: <warn>  [1606879638.4747] device (wlp5s0u2): Activation: (wifi) asking for new secrets
Dec 01 21:27:21 ignyter NetworkManager[659]: <info>  [1606879641.9558] device (wlp5s0u2): supplicant interface state: associating -> disconnected
Dec 01 21:27:21 ignyter NetworkManager[659]: <info>  [1606879641.9559] device (p2p-dev-wlp5s0u2): supplicant management interface state: associating -> disconnected
Dec 01 21:27:21 ignyter NetworkManager[659]: <warn>  [1606879641.9648] device (wlp5s0u2): no secrets: User canceled the secrets request.
Dec 01 21:27:21 ignyter NetworkManager[659]: <info>  [1606879641.9648] device (wlp5s0u2): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Dec 01 21:27:21 ignyter NetworkManager[659]: <info>  [1606879641.9656] manager: NetworkManager state is now CONNECTED_LOCAL
Dec 01 21:27:21 ignyter NetworkManager[659]: <warn>  [1606879641.9668] device (wlp5s0u2): Activation: failed for connection 'myssid'
Dec 01 21:27:21 ignyter NetworkManager[659]: <info>  [1606879641.9671] device (wlp5s0u2): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
```
It just continues prompting for the PSK over and over and never connects.  Is there something obvious I'm missing?

RinCat, 2020-12-02:
Hi, set `echo 6 > /proc/net/rtl88x2bu/log_level` and see `dmesg`.
Is there any `new beacon occur!!` in the log?

jsermer, 2020-12-03:
Attached the log, but the only 'beacon' messages I see are:

```
[  +0.000005] RTW: OnBeacon: beacon keys ready
[  +0.000007] RTW: OnBeacon: beacon keys ready
[  +0.000006] RTW: OnBeacon: beacon keys ready
```

[dmesg.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5632864/dmesg.txt)

RinCat, 2020-12-03:
RTW: assoc reject, status code: 40
Invalid information element (doesn't follow 802.11 standard)
https://support.google.com/chrome/a/answer/7172038

I guess some of your configuration does not compatible with your wifi.

jsermer, 2020-12-03:
Hmm, my NetworkManager config is bare bones (/etc/NetworkManager/NetworkManager.conf is effectively blank) and my wifi config file is:

```
[connection]
id=myssid
uuid=39a38116-0be8-48a7-9ffc-8a378c9330c5
type=wifi
interface-name=wlp5s0u2
permissions=

[wifi]
bssid=myssid_bssid_mac_address
mac-address-blacklist=
mode=infrastructure
ssid=myssid

[wifi-security]
key-mgmt=wpa-psk
psk=mypsk

[ipv4]
dns-search=
method=auto

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=disabled

[proxy]
```

RinCat, 2020-12-03:
Try change your wifi router configuration and see if it works.

jsermer, 2020-12-03:
It works using the same basic config options (ssid, bssid, and psk) manually with wpa_supplicant directly but not using those same options with NetworkManager (I'm actually sending this message over the wifi adapter manually configured using wpa_supplicant command line):

```
wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant-wlp5s0u2.conf -iwlp5s0u2
```

RinCat, 2020-12-03:
I do use NetworkManager and iwd, and they both works fine for me. So I don't know what special reasons may causes NM to not work properly. Your wifi authentication is successful, but failed at association without any useful info.

jsermer, 2020-12-03:
Switching the wifi backend in NetworkManager to iwd, restarting NetworkManager, removing existing 'known wifi networks', killing the existing wpa_supplicant, and re-attempting a connection using the wifi settings in gnome control center seems to have made everything work as expected.

jsermer, 2020-12-05:
Still unsure why wpa_supplicant works natively, but not via NetworkManager using the default wifi.backend of wpa_supplicant.  I can report however, that changing the backend to iwd seems to work pretty well.

/etc/NetworkManager/conf.d/wifi_backend.conf:
``` 
[device]
wifi.backend=iwd
```

I will also note, initially the 2.4ghz band was preferred (2.4ghz signals travel better through walls), but since this is a desktop and doesn't move, I wanted to force the adapter to connect to my 80mhz 5ghz band.  iwd made it pretty simple to influence the 5ghz preference by adjusting the following:

/etc/iwd/main.conf
```
[Rank]
BandModifier5Ghz=2.0
```

While not technically 'solved', you are free to close this issue.