RinCat/RTL88x2BU-Linux-Driver #62: AP Mode: Unable to achieve VHT operation (867mbps)

jbd1986, 2020-12-16:
I'm unable to achieve a VHT connection in ap mode, unless I make a modification to core/rtw_ap.c:

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/fac54b523d48bcba2f8f0d320e50952bfbf2b26e/core/rtw_ap.c#L2387

We are in MLME_IS_AP mode, not MLME_IS_MESH mode at this point.  I'm not sure what the correct solution is, but if I simply stop checking for MLME_IS_MESH mode here, I am able to connect at 867mbps (though iwconfig shows 802.11bgn instead of 802.11AC).

Do you have any insight here?

jbd1986, 2020-12-16:
As for the incorrect iwconfig display, it looks like it probably comes from:

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/fac54b523d48bcba2f8f0d320e50952bfbf2b26e/core/rtw_ap.c#L2013

That function from core/rtw_ieee80211.c seems to only be called in this one place (and should possibly be removed, or at least updated to support 80211AC/vht in a meaningful way).  Replacing that function call with the following resolves the iwconfig display issue:
	network_type = WIRELESS_11_5AC;
	rtw_set_supported_rate(pbss_network->SupportedRates, network_type);

jbd1986, 2020-12-16:
As an update, the following solution resolves the issue for me:

`diff --git a/core/rtw_ap.c b/core/rtw_ap.c
index 5949ad0..299f58d 100644
--- a/core/rtw_ap.c
+++ b/core/rtw_ap.c
@@ -2010,7 +2010,9 @@ int rtw_check_beacon_data(_adapter *padapter, u8 *pbuf,  int len)
 	pbss_network->Configuration.DSConfig = channel;
 
 	/*	support rate ie & ext support ie & IElen & SupportedRates	*/
-	network_type = rtw_update_rate_bymode(pbss_network, pregistrypriv->wireless_mode);
+	//network_type = rtw_update_rate_bymode(pbss_network, pregistrypriv->wireless_mode);
+	network_type = WIRELESS_11_5AC;
+	rtw_set_supported_rate(pbss_network->SupportedRates, network_type);
 
 	/* parsing ERP_IE */
 	p = rtw_get_ie(ie + _BEACON_IE_OFFSET_, _ERPINFO_IE_, &ie_len, (pbss_network->IELength - _BEACON_IE_OFFSET_));
@@ -2376,6 +2378,9 @@ int rtw_check_beacon_data(_adapter *padapter, u8 *pbuf,  int len)
 		&& is_supported_vht(pregistrypriv->wireless_mode)
 		&& (!rfctl->country_ent || COUNTRY_CHPLAN_EN_11AC(rfctl->country_ent))
 	) {
+        /* enable the vht_option so that VHT can be toggled as needed in other functions */
+        pmlmepriv->vhtpriv.vht_option = _TRUE;
+
 		/* Parsing VHT CAP IE */
 		p = rtw_get_ie(ie + _BEACON_IE_OFFSET_, EID_VHTCapability, &ie_len, (pbss_network->IELength - _BEACON_IE_OFFSET_));
 		if (p && ie_len > 0)
@@ -2388,7 +2393,6 @@ int rtw_check_beacon_data(_adapter *padapter, u8 *pbuf,  int len)
 		) {
 			rtw_check_for_vht20(padapter, ie + _BEACON_IE_OFFSET_, pbss_network->IELength - _BEACON_IE_OFFSET_);
 			pmlmepriv->ori_vht_en = 1;
-			pmlmepriv->vhtpriv.vht_option = _TRUE;
 		} else if (REGSTY_IS_11AC_AUTO(pregistrypriv)) {
 			rtw_vht_ies_detach(padapter, pbss_network);
 			rtw_vht_ies_attach(padapter, pbss_network);
`

RinCat, 2020-12-17:
Maybe you can try `modprobe 88x2bu vht_enable = 2`.

jbd1986, 2020-12-17:
@RinCat, the module param 'rtw_vht_enable=2' did enable VHT speeds at 867mbps (though iwconfig was still incorrectly showing 802.11bgn).  Is that the only way that should be achievable?

RinCat, 2020-12-17:
@jbd1986 I don't know since I don't have their internal documents, I can only guess why their logic is set up like this. 
Set `vht_enable = 2` is the only way let the `} else if (REGSTY_IS_11AC_AUTO(pregistrypriv)) {` to be true and try to enable vht.

IlyaMZP, 2021-01-12:
I had tested that VHT actually works with `rtw_vht_enable=2` flag and got upload speeds over 150mbps with [this config](https://mzp.icu/files/hostapd.conf).
But my download speeds sunk below 100mbps for some reason. It was probably due to poor power supply I have. Connecting it to my PC improved speeds, but I still prefer to leave VHT disabled for AP mode.

jbd1986, 2021-03-15:
FYI, we're currently having an issue with both Miracast and AP mode, where we cannot achieve 'wireless N mode' when rtw_vht_enable=2 is set.  This is required to work for compliance testing... so I'm not really sure how to move forward at the moment.

RinCat, 2021-03-15:
@jbd1986 VHT is a feature in 802.11ac, it does not work if you set it to 802.11n mode. 802.11n only support HT.

jbd1986, 2021-03-16:
@RinCat yes understood.  The issue is that we want to be able to support 802.11ac w/ VHT as well as 802.11n without VHT. 

Currently I can only support one or the other.

RinCat, 2021-03-17:
@jbd1986 you can remove and reload the module to do that. By default `rtw_vht_enable=1` which will enable VHT if available, but it usually not work. This may be a bug in the original code.

jbd1986, 2021-03-22:
@RinCat We may end up using that as a workaround, I had already discussed that possibility with my team.  We'd prefer a proper solution in the driver of course :)