RinCat/RTL88x2BU-Linux-Driver #63: Build fails with kernel 5.10

lbroman, 2020-12-16:
Building the module with kernel 5.10 fails with the error
`
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.1-051001-generic/build M=/home/user/workspaces/GIT/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.1-051001-generic'
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/home/user/workspaces/GIT/RTL88x2BU-Linux-Driver/88x2bu.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1709: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.1-051001-generic'
make: *** [Makefile:2378: modules] Error 2
`

A bug related to this has also been reported on launchpad: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131

RinCat, 2020-12-17:
https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/58

lbroman, 2020-12-17:
Sorry I didn't realise that bug was the same issue. Closing this.

closed 2020-12-17 by lbroman