RinCat/RTL88x2BU-Linux-Driver #65: Building not working on Linux Kernel 5.10.1

gagallo7, 2020-12-20:
First of all, thanks for sharing this driver updates.

So, I purchased a EDUP EP-1689, commercially named `EDUP 1300Mbps Mini USB3.0 Wifi Adapter`, but I had some troubles compiling its driver with the package provided Linux headers in Debian 10.
Then I downloaded the mainline kernel source, compiled it myself, and used it as source to your driver version.

But some building errors happened related with [`cfg80211_ch_switch_started_notify`](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/include/net/cfg80211.h#n7576) ioctl.
A new parameter, `quiet` was added, with the following documentation:
```
@quiet: whether or not immediate quiet was requested by the AP
```

I added this argument with the value `true`. Then the driver built successfully.

With the module installed, I had some troubles with auth loop, then I stumbled upon this [comment](https://aur.archlinux.org/packages/rtl88x2bu-dkms-git/#comment-777552). Using the tips, I was able to get the adapter working as a charm on my setup.

I opened a [PR](https://github.com/RinCat/RTL88x2BU-Linux-Driver/pull/64) related with my fix, please review it if possible, as I assumed that the new value should always be `true`.

RinCat, 2020-12-20:
Thanks, will check tomorrow.

RinCat, 2020-12-21:
Merged, thanks.

RinCat, 2020-12-21:
Well, there are 2 things. First the quiet should be default false. Second, kernel change is not in the 5.10 branch yet. I temporarily put it into 5.11. So you have to patch it by your own at the moment.

gagallo7, 2020-12-21:
Great. Thanks for the review!

closed 2020-12-21 by gagallo7