RinCat/RTL88x2BU-Linux-Driver #66: Can't compile

sitaro, 2020-12-30:
Hello,
on compile I got this Error:

/root/RTL88x2BU-Linux-Driver/os_dep/linux/os_intfs.c:1715:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,

System is CentOS 8.3.2011
Compiler is gcc-8.3.1

Thanks

RinCat, 2020-12-30:
Hi, what is the kernel version?

iplasma, 2020-12-31:
Have the same problem on CentOS 8 wit kernel 4.18.0-240.el8.x86_64. 

Looks like problem is with backport in the kernel. see discussion [here](https://forums.opensuse.org/showthread.php/535241-I-can-t-make-rtl8822bu). I made the changes in the [patch](https://build.opensuse.org/package/view_file/home:Sauerland/rtl8822bu/Leap151.patch?expand=1). There is another place in rtw_android. there is a similar problem. After that, the [driver](https://github.com/cilynx/rtl88x2BU_WiFi_linux_v5.3.1_27678.20180430_COEX20180427-5959) compiles fine but crashed after the usb wireless dongle is plugged in. Haven't tried the driver here yet.

RinCat, 2020-12-31:
Seems this https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/9f94efd6b0ea7bce67781ba4a424d4d7befd2cd1, you need change 4.19 to 4.18 since this is backport.

And I cannot put this in the driver because this is distribution backport, you need find out what they did and change the preprocess kernel version.

iplasma, 2020-12-31:
Thanks, try your suggestion and the driver here compiles but still crashes. Switch to 5.10.4-1.el8.elrepo.x86_64 kernel from elrepo and now it works.

bagustris, 2021-04-07:
hi @iplasma, what configuration you used to compile with elrepo kernel?  I tried, but it always fails with unavailability of kernel headers (which is not possible to install with elrepo, only using original distribution).

CGarces, 2021-04-07:
@RinCat 
Can you please check this? 
https://github.com/Mange/rtl8192eu-linux-driver/pull/232

It use RHEL_RELEASE_CODE to not break compilation on mainline kernels

RinCat, 2021-04-07:
@CGarces I know that exists, but I prefer not add any distribution specific patches, as I can't check if each kernel in each distribution backports a certain patch. The easiest way is to manually modify the kernel version macro if you run into this， `KERNEL_VERSION(4, 19, 0)` to `KERNEL_VERSION(4, 18, 0)`.

mrlee14, 2021-07-28:
Success on Centos 8 (stream), as it fails to compile on default kernel what I done, first enabled elrepo repository, next installed kernel 5.4.135-1, make, make install, reboot....it works very nice. Thank you!