RinCat/RTL88x2BU-Linux-Driver #67: Wifi-Password not accepted

guckuck, 2021-01-03:
I cannot connect to my wifi-network. The password is not accepted. The password is correct.

I can connect to a wifi-network without password (for example Telekom-FON).

My System:
TP-Link Archer T3U Plus
lubuntu 18.04 (4.15.0-128-generic) 32bit
Speedport Pro Plus (German Telekom) (3 wifis: 2,4 GHz, 5 Ghz with same ssid, TELEKOM-FON).
```

syslog:
Jan  3 23:49:13 tux kernel: [ 1223.816135] RTW: unknown vendor specific information element ignored (vendor OUI 50:6f:9a len=7)
Jan  3 23:49:13 tux kernel: [ 1223.816138] RTW: IEEE 802.11 element parse ignored unknown element (id=108 elen=2)
Jan  3 23:49:13 tux kernel: [ 1223.816141] RTW: IEEE 802.11 element parse ignored unknown element (id=107 elen=7)
Jan  3 23:49:13 tux kernel: [ 1223.904815] RTW: [HALMAC][TRACE]cfg_usb_rx_agg_88xx ===>
Jan  3 23:49:13 tux kernel: [ 1223.905686] RTW: [HALMAC][TRACE]cfg_usb_rx_agg_88xx <===
Jan  3 23:49:13 tux kernel: [ 1223.905812] RTW: [HALMAC][TRACE]cfg_csi_rate_88xx ===>
Jan  3 23:49:13 tux kernel: [ 1223.906440] RTW: rtl8822b_fillh2ccmd(wlx984827a71691): id=0x42 buf= 0x00 0x00 0x36 0x40 0x00 0x00 0x00
Jan  3 23:49:13 tux kernel: [ 1224.020849] RTW: IEEE 802.11 element parse ignored unknown element (id=7 elen=52)
Jan  3 23:49:13 tux kernel: [ 1224.020865] RTW: IEEE 802.11 element parse ignored unknown element (id=32 elen=1)
Jan  3 23:49:13 tux kernel: [ 1224.020867] RTW: IEEE 802.11 element parse ignored unknown element (id=195 elen=4)
Jan  3 23:49:13 tux kernel: [ 1224.020870] RTW: unknown vendor specific information element ignored (vendor OUI 00:26:86 len=34)
Jan  3 23:49:13 tux kernel: [ 1224.020871] RTW: Unknown Broadcom information element ignored (type=4 len=5)
Jan  3 23:49:13 tux kernel: [ 1224.020873] RTW: unknown vendor specific information element ignored (vendor OUI 00:26:86 len=6)
Jan  3 23:49:13 tux kernel: [ 1224.020875] RTW: IEEE 802.11 element parse ignored unknown element (id=127 elen=8)
Jan  3 23:49:13 tux kernel: [ 1224.020876] RTW: unknown vendor specific information element ignored (vendor OUI 50:6f:9a len=7)
Jan  3 23:49:13 tux kernel: [ 1224.020878] RTW: IEEE 802.11 element parse ignored unknown element (id=108 elen=2)
Jan  3 23:49:13 tux kernel: [ 1224.020879] RTW: IEEE 802.11 element parse ignored unknown element (id=107 elen=7)
Jan  3 23:49:13 tux kernel: [ 1224.225718] RTW: IEEE 802.11 element parse ignore
```

I hope, somebody can help me.

Best regards
Kay

RinCat, 2021-01-05:
Not enough logs, unknown element are expected.

MahouShoujoMivutilde, 2021-01-09:
Do you use networkmanager with wpa_supplicant?

Had similar experience with comfast cf 913ac v2 today.

Switching to iwd (as suggested [here](https://askubuntu.com/a/1271252)) helped.

guckuck, 2021-01-09:
> Do you use networkmanager with wpa_supplicant?
> 
> Had similar experience with comfast cf 913ac v2 today.
> 
> Switching to iwd (as suggested [here](https://askubuntu.com/a/1271252)) helped.

Thanks, I will test that. Sadly there is no package for Ubuntu 18.4 (32 bit), so I have to compile it for myself.

guckuck, 2021-01-09:
> Not enough logs, unknown element are expected.

This log shows the first try to connect to my wifi ("Funkloch"). After that fails, it connect automatically with the T-Online FON-Hotspot (without key). 

```Jan  9 16:17:54 tux avahi-daemon[748]: Interface wlx984827a71691.IPv4 no longer relevant for mDNS.
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1379] device (wlx984827a71691): Activation: (wifi) access point 'Funkloch 6' has security, but secrets are required.
Jan  9 16:17:54 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-DISCONNECTED bssid=e6:08:4f:8f:53:63 reason=3 locally_generated=1
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1380] device (wlx984827a71691): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Jan  9 16:17:54 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan  9 16:17:54 tux NetworkManager[752]: <warn>  [1610205474.1431] sup-iface[0x1d92508,wlx984827a71691]: connection disconnected (reason -3)
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1432] device (wlx984827a71691): supplicant interface state: completed -> disconnected
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1590] device (wlx984827a71691): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1600] device (wlx984827a71691): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1718] device (wlx984827a71691): Activation: (wifi) connection 'Funkloch 6' has security, and secrets exist.  No new secrets needed.
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1718] Config: added 'ssid' value 'Funkloch'
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1719] Config: added 'scan_ssid' value '1'
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1719] Config: added 'bgscan' value 'simple:30:-80:86400'
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1719] Config: added 'key_mgmt' value 'WPA-PSK'
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1719] Config: added 'psk' value '<hidden>'
Jan  9 16:17:54 tux wpa_supplicant[751]: wlx984827a71691: Trying to associate with f8:08:4f:8f:53:63 (SSID='Funkloch' freq=5180 MHz)
Jan  9 16:17:54 tux kernel: [ 2554.745053] RTW: rtw_set_802_11_connect(wlx984827a71691)  fw_state=0x00000008
Jan  9 16:17:54 tux NetworkManager[752]: <info>  [1610205474.1890] device (wlx984827a71691): supplicant interface state: disconnected -> associating
Jan  9 16:17:57 tux nautilus[1681]: Attempting to read the recently used resources file at '/home/papa/.local/share/recently-used.xbel', but the parser failed: Datei »/home/papa/.local/share/recently-used.xbel« konnte nicht geöffnet werden: Keine Berechtigung.
Jan  9 16:17:57 tux gedit[2979]: Attempting to read the recently used resources file at '/home/papa/.local/share/recently-used.xbel', but the parser failed: Datei »/home/papa/.local/share/recently-used.xbel« konnte nicht geöffnet werden: Keine Berechtigung.
Jan  9 16:18:00 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-ASSOC-REJECT status_code=1
Jan  9 16:18:00 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan  9 16:18:00 tux NetworkManager[752]: <info>  [1610205480.4603] device (wlx984827a71691): supplicant interface state: associating -> disconnected
Jan  9 16:18:00 tux NetworkManager[752]: <warn>  [1610205480.4603] sup-iface[0x1d92508,wlx984827a71691]: connection disconnected (reason 1)
Jan  9 16:18:00 tux kernel: [ 2561.020027] RTW: WARN _beamforming_leave: f8:08:4f:8f:53:63 is neither beamforming ee or er!!
Jan  9 16:18:00 tux NetworkManager[752]: <info>  [1610205480.5648] device (wlx984827a71691): supplicant interface state: disconnected -> scanning
Jan  9 16:18:01 tux wpa_supplicant[751]: wlx984827a71691: Trying to associate with f8:08:4f:8f:53:64 (SSID='Funkloch' freq=2462 MHz)
Jan  9 16:18:01 tux kernel: [ 2561.788170] RTW: rtw_set_802_11_connect(wlx984827a71691)  fw_state=0x00000008
Jan  9 16:18:01 tux NetworkManager[752]: <info>  [1610205481.2365] device (wlx984827a71691): supplicant interface state: scanning -> associating
Jan  9 16:18:07 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-ASSOC-REJECT status_code=1
Jan  9 16:18:07 tux wpa_supplicant[751]: wlx984827a71691: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jan  9 16:18:07 tux NetworkManager[752]: <info>  [1610205487.3683] device (wlx984827a71691): supplicant interface state: associating -> disconnected
Jan  9 16:18:07 tux NetworkManager[752]: <warn>  [1610205487.3683] sup-iface[0x1d92508,wlx984827a71691]: connection disconnected (reason 1)
Jan  9 16:18:07 tux kernel: [ 2567.928166] RTW: WARN _beamforming_leave: f8:08:4f:8f:53:64 is neither beamforming ee or er!!
Jan  9 16:18:07 tux NetworkManager[752]: <info>  [1610205487.4728] device (wlx984827a71691): supplicant interface state: disconnected -> scanning
Jan  9 16:18:11 tux wpa_supplicant[751]: wlx984827a71691: Trying to associate with f8:08:4f:8f:53:63 (SSID='Funkloch' freq=5180 MHz)
Jan  9 16:18:11 tux kernel: [ 2572.212532] RTW: rtw_set_802_11_connect(wlx984827a71691)  fw_state=0x00000008
Jan  9 16:18:11 tux NetworkManager[752]: <info>  [1610205491.6578] device (wlx984827a71691): supplicant interface state: scanning -> associating
[... cut ...]
Jan  9 16:18:33 tux nm-dispatcher: req:2 'connectivity-change': start running ordered scripts...
Jan  9 16:18:33 tux systemd[1]: Reloaded OpenBSD Secure Shell server.
Jan  9 16:18:34 tux avahi-daemon[748]: Joining mDNS multicast group on interface wlx984827a71691.IPv6 with address fe80::dadb:3b74:829e:88f9.
Jan  9 16:18:34 tux avahi-daemon[748]: New relevant interface wlx984827a71691.IPv6 for mDNS.
Jan  9 16:18:34 tux avahi-daemon[748]: Registering new address record for fe80::dadb:3b74:829e:88f9 on wlx984827a71691.*.
Jan  9 16:18:36 tux whoopsie[1012]: [16:18:36] online```

RinCat, 2021-01-09:
Try set `echo 5 > /proc/net/rtl88x2bu/log_level` and check logs. You may wanna try iwd as mentioned.

guckuck, 2021-01-09:
So, I updated my system to Ubuntu 20.10 (64 bit). I had no luck with iwd. Same problem. Cannot connect to my wifi. The fon-hotsport (Telekom_FON) also won't work because no captive portal is shown. But I can connect to this wifi.

Maybe important: I use the same SSID for the 2,4 und 5 GHz network.

I created a new log.

[log.txt](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5791949/log.txt)

RinCat, 2021-01-10:
What's your WiFi model and configuration? The driver think the WiFi beacon frame is invalid.

guckuck, 2021-01-10:
My router is a Speedport Pro Plus (very new model). As attachment my configuration. I hope, you understand it. It is in german. Do you need more information?

[image]
[image]

RinCat, 2021-01-10:
Maybe try change both to 802.11n/ac , the router send beacons said the size is 80, but size only 26. It could be bug in the router.

guckuck, 2021-01-16:
I can not change the 2,4 GHz-Network to n/ac. I have the following options: b/g/n; g/n; b/g

I will now test some other options. Maybe I get correct beacons.

guckuck, 2021-01-16:
I had no luck. I will buy another wifi-stick for this computer. Thanks for your support and your work.