RinCat/RTL88x2BU-Linux-Driver #68: RTL8812BU add device ID to whitelist

scottsteen, 2021-01-08:
The installation through `dkms` was seemingly successful, but Ubuntu doesn't display the WiFi page in the settings window & `nmcli dev wifi` produces no results. I hope this is simply a whitelisting issue.

Adapter info:
Maxesla Wifi Adapter - AC1200 USB Wifi Dongle
Link: https://www.amazon.co.uk/gp/product/B08GS4PX13
```
$ lsusb 
Bus 003 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. USB3.0 802.11ac 1200M Adapter

$ dkms status rtl88x2bu
rtl88x2bu, git, 5.8.0-34-generic, x86_64: installed
```

System info:
```
$ uname -a
Linux work 5.8.0-36-generic #40~20.04.1-Ubuntu SMP Wed Jan 6 10:15:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

I'm happy to submit a PR (given pointers!) or test changes if that'll help.

CGarces, 2021-01-08:
Hi.

The device 0bda:b812 already exists on the source code.

https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/fac54b523d48bcba2f8f0d320e50952bfbf2b26e/os_dep/linux/usb_intf.c#L238

Please take a look over previous issues and check if match with your current problem.

https://github.com/RinCat/RTL88x2BU-Linux-Driver/search?q=b812&type=issues

scottsteen, 2021-01-08:
Thanks for the quick response! Apologies, I didn't realise my adapter was already supported.

I can't see anything in the existing issues besides https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/21, but it has little to no detail.

I'll have an investigation into my system set up to see if there's anything obvious going wrong.

closed 2021-01-08 by scottsteen