RinCat/RTL88x2BU-Linux-Driver #69: Hotspot is Really Slow (High latency on connections)

TheDiscordian, 2021-01-15:
Hello, I've been trying to solve this issue for about a week. Our wireless device works great when acting as a client, but when we make a Hotspot, connections to the local HTTP server are very slow, taking well over 10 seconds to respond to requests.

dmesg log: https://pastebin.com/EtqKDGJF
NetworkManager log: https://gist.github.com/TheDiscordian/15c813d67905673213025b37c36cbc05

The device is a Raspberry Pi4 running a USB wireless card (Cudy WU1400). The issue persists even on runs where we don't get the undervolting issue. The card is attached to a USB extender, however directly connecting it to the Pi has no improvement either.

I've tried investigating this issue by trying various fixes I've read for other devices, like disabling power save, however we've observed no changes.

RinCat, 2021-01-15:
You may try this, but I didn't test it. https://blogs.gnome.org/dcbw/2016/05/16/networkmanager-and-wifi-scans/

TheDiscordian, 2021-01-15:
Thank you for your incredibly fast response (I went to bed after I posted, sorry!). I've read that, and I wasn't sure how to apply it to a Hotspot configuration, that seems to be good for configuring a client to connect to a specific AP, no? If I'm totally wrong here, I'd appreciate an example if possible, as I'm not sure what BSSID I'd be specifying as a hotspot.

RinCat, 2021-01-15:
Maybe you can try disable NM and only use wpa_supplicant?