RinCat/RTL88x2BU-Linux-Driver #70: ERROR (dkms apport): binary package for rtl88x2bu: git not found

jackyleehong, 2021-01-16:
After i run dkms autoinstall, it show error:

'make' -j8 KVER=5.8.0-38-generic KSRC=/lib/modules/5.8.0-38-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: git not found
Error! Bad return status for module build on kernel: 5.8.0-38-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/git/build/make.log for more information.

RinCat, 2021-01-16:
git clone to your local, and try manual build. And could you upload log.

jackyleehong, 2021-01-17:
When i run the make command, it shown up this error instead.
[image]

RinCat, 2021-01-17:
Your kernel headers had some issue, is it the same version as you current running? Maybe you can try reinstall linux-headers.

jackyleehong, 2021-01-24:
Thanks bro. Issue have been solved, because of the nvidiabl build issue.