RinCat/RTL88x2BU-Linux-Driver #71: Cannot make repository

Jayjayd123, 2021-01-17:
Whenever I type "make" when in the driver directory, I get this result:

make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.4.83-v7l+/build M=/home/pi/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-v7l+'
Makefile:671: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.83-v7l+'
make: *** [Makefile:2378: modules] Error 2

any help?

RinCat, 2021-01-18:
I guess you need set `ARCH=arm64`

joseluisdelara, 2021-02-12:
I am having the same issue. Did ARCH=arm64 fix the problem? if so, where did you perform the change?
Thanks a lot for your help

eschwelgin, 2021-03-07:
Similar issue on raspi running ubuntu 20.10 server, ARCH=arm64 throwing a different error 

`ubuntu@ubuntu:/lib/modules/5.8.0-1016-raspi/kernel/drivers$ sudo git clone https://github.com/RinCat/RTL88x2BU-Linux-Driver.git
Cloning into 'RTL88x2BU-Linux-Driver'...
remote: Enumerating objects: 721, done.
remote: Counting objects: 100% (721/721), done.
remote: Compressing objects: 100% (581/581), done.
remote: Total 2091 (delta 283), reused 255 (delta 138), pack-reused 1370
Receiving objects: 100% (2091/2091), 9.42 MiB | 7.07 MiB/s, done.
Resolving deltas: 100% (1059/1059), done.
ubuntu@ubuntu:/lib/modules/5.8.0-1016-raspi/kernel/drivers$ cd RTL88x2BU-Linux-Driver/
ubuntu@ubuntu:/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver$ sudo make ARCH=arm64
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.8.0-1016-raspi/build M=/usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-1016-raspi'
  CC [M]  /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
In file included from ./include/linux/types.h:6,
                 from /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver/include/basic_types.h:75,
                 from /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver/include/drv_types.h:26,
                 from /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver/core/rtw_cmd.c:17:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:286: /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1786: /usr/lib/modules/5.8.0-1016-raspi/kernel/drivers/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-1016-raspi'
make: *** [Makefile:2378: modules] Error 2
`

RinCat, 2021-03-07:
@eschwelgin try `make ARCH=aarch64` or `make ARCH=arm`? 
I'm not sure about your kernel configuration, and I don't have rpi to try it, but some people report it works.

https://www.raspberrypi.org/documentation/linux/kernel/building.md

eschwelgin, 2021-03-07:
> @eschwelgin try `make ARCH=aarch64` or `make ARCH=arm`?
> I'm not sure about your kernel configuration, and I don't have rpi to try it, but some people report it works.
> 
> https://www.raspberrypi.org/documentation/linux/kernel/building.md

Thank you for the quick reply

This worked for me and im up and running. Thanks!

MrZell, 2021-03-22:
I tried compiling this for a Pi 3b running Debian Stretch (4.14.79-v7+) and was getting the "No such file or directory" message. Took a little bit of digging (I'm not familiar with compiling drivers), but all I had to do to get it working was add:
`CONFIG_PLATFORM_ARM_RPI = y` to the "Platform Related" section of the Makefile and then add the below code block somewhere alongside the other `ifeq ($(CONFIG_PLATFORM_...), y)` blocks (I put it around 1300 lines in just above the `ifeq ($(CONFIG_PLATFORM_I386_PC), y)` if block). After doing that, make ran just fine without any errors. This is the method I have seen within other Makefiles in the past for the Raspberry Pi (3b specifically).

`ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)`
`EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN`
`EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT`
`ARCH := arm`
`KVER := $(shell uname -r)`
`KSRC := /lib/modules/$(KVER)/build`
`MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/`
`INSTALL_PREFIX :=`
`STAGINGMODDIR := /lib/modules/$(KVER)/kernel/drivers/staging`
`endif`

Upon further investigation, just running `make ARCH=arm` worked as well without modifying the Makefile.

joseluisdelara, 2021-03-25:
Thanks a lot to taking the time for this explanation.

I ll make the try!

 

José Luis

SUPERustam, 2021-04-05:
same problem:

make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.9.16-1-MANJARO/build M=/home/superustam/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make: *** [Makefile:2378: modules] Error 2

RinCat, 2021-04-05:
@SUPERustam try `ARCH=aarch64` ?

SUPERustam, 2021-04-06:
make ARCH=aarch64                      
```bash
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.9.16-1-MANJARO/build M=/home/superustam/RTL8812BU  modules
make[1]: Entering directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make: *** [Makefile:2288: modules] Error 2
```

SUPERustam, 2021-04-06:
@RinCat not work(

RinCat, 2021-04-06:
@SUPERustam is there anything in `/lib/modules/5.9.16-1-MANJARO/build` ?

SUPERustam, 2021-04-06:
@RinCat Yeah! vmlinux file
[image]

RinCat, 2021-04-06:
@SUPERustam that's not what it should. It should have a lot of files for build modules, did you install something like `linux-headers`?

SUPERustam, 2021-04-06:
No, i didn't install something like `linux-headers`. Should i install linux-headers? Via `pacman` or another way?

RinCat, 2021-04-06:
@SUPERustam install the same way as your kernel. So if you are using `linux` package, then install `linux-headers` from `pacman`.

SUPERustam, 2021-04-07:
```bash
sudo pacman -S linux-headers        
[sudo] password for superustam: 
:: There are 11 providers available for linux-headers:
:: Repository core
   1) linux414-headers  2) linux419-headers  3) linux44-headers
   4) linux49-headers  5) linux510-headers  6) linux511-headers
   7) linux512-headers  8) linux54-headers  9) linux59-headers
:: Repository community
   10) linux54-rt-headers  11) linux59-rt-headers

Enter a number (default=1): 
```
@RinCat which provider i should choose?

Computer setup (if it's matter)):
First disk: 5.9.16-1-MANJARO 
Second disk: Windows 10
CPU: AMD Ryzen 5 2400g
GPU: AMD Radeon Vega 11

RinCat, 2021-04-07:
You are on x86_64? Since you are using 5.9.16-1-MANJARO, go use linux59-headers

SUPERustam, 2021-04-07:
[image]
 
Error:
```bash 
sudo pacman -S linux-headers                                      
[sudo] password for superustam: 
:: There are 11 providers available for linux-headers:
:: Repository core
   1) linux414-headers  2) linux419-headers  3) linux44-headers  4) linux49-headers  5) linux510-headers
   6) linux511-headers  7) linux512-headers  8) linux54-headers  9) linux59-headers
:: Repository community
   10) linux54-rt-headers  11) linux59-rt-headers

Enter a number (default=1): 9
resolving dependencies...
looking for conflicting packages...

Packages (1) linux59-headers-5.9.16-1

Total Download Size:    9.68 MiB
Total Installed Size:  48.27 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 linux59-headers-5.9.16-1-x86_64          9.7 MiB   346 KiB/s 00:29 [######################################] 100%
(1/1) checking keys in keyring                                      [######################################] 100%
(1/1) checking package integrity                                    [######################################] 100%
(1/1) loading package files                                         [######################################] 100%
(1/1) checking for file conflicts                                   [######################################] 100%
(1/1) checking available disk space                                 [######################################] 100%
:: Processing package changes...
(1/1) installing linux59-headers                                    [######################################] 100%
:: Running post-transaction hooks...
(1/3) Arming ConditionNeedsUpdate...
(2/3) Updating module dependencies...
(3/3) Install DKMS modules
 -k 5.9.16-1-MANJAROo-depmod -m rtl88x2bu -v 5.6.1
Error! Bad return status for module build on kernel: 5.9.16-1-MANJARO (x86_64)
/build/make.log for more information.
 -k 5.9.16-1-MANJARO' returned 10epmod -m rtl88x2bu -v 5.6.1
==> dkms install --no-depmod -m 8812au -v 5.6.4.2_35491.20191025 -k 5.9.16-1-MANJARO
^C
Interrupt signal received
error: command terminated by signal 2: Interrupt

```

RinCat, 2021-04-07:
@SUPERustam Driver version 5.6.1? That's too old for 5.9 kernel, update it first.

SUPERustam, 2021-04-07:
I found two day ago this https://github.com/fastoe/RTL8812BU/tree/v5.6.1 and today it right run my adapter.
Thank You @RinCat for help me! I really grateful for advices you give me in this issue :heart:.