RinCat/RTL88x2BU-Linux-Driver #72: kernel 5.10.7-100.fc32.x86_64 (x86_64) make error

zerobabble, 2021-01-18:
Hi, 

getting the following build error with the latest kernel... log attached... 

[make.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5827828/make.log)

RinCat, 2021-01-18:
You need update your local driver version.
Reinstall if you use AUR.

If you use the script in the README, run
```
cd /usr/src/rtl88x2bu-git
git pull
```

zerobabble, 2021-01-18:
great and thank you for the fast reply... pulled in the source changes and now am getting the following error... 

[make.log](https://github.com/RinCat/RTL88x2BU-Linux-Driver/files/5828315/make.log)

`In file included from /usr/src/rtl88x2bu-5.6.1/include/drv_types.h:79,
from /usr/src/rtl88x2bu-5.6.1/core/rtw_security.c:17:
/usr/src/rtl88x2bu-5.6.1/include/rtw_security.h:252:8: error: redefinition of ‘struct sha256_state’
252 | struct sha256_state {
|        ^~~~~~~~~~~~`

RinCat, 2021-01-18:
emmm, you had 2 same driver? The first driver show using git, the second log shows 5.6.1.
Remove `/usr/src/rtl88x2bu-5.6.1/`

zerobabble, 2021-01-18:
cleaned up dkms and did a clean checkout... all sorted, thank you!

closed 2021-01-18 by zerobabble