RinCat/RTL88x2BU-Linux-Driver #75: Add pre-requisites to the readme

lazybearcreations, 2021-02-18:
My `make` was failing and I couldn't work out why...
After some googleing I found out that I hadn't installed `build-essential`
Can you add that to the readme so n00bs [like myself] don't get tripped up with that simple fix.

CGarces, 2021-02-18:
Feel free to send a pull request, it's very easy. You don't need any tool, can be done using the web interface.
You can use [this readme.md](https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/README.md)  as reference for markdown syntaxis.

lazybearcreations, 2021-02-18:
> Feel free to send a pull request, it's very easy. You don't need any tool, can be done using the web interface.
> You can use [this readme.md](https://github.com/aircrack-ng/rtl8812au/blob/v5.6.4.2/README.md) as reference for markdown syntaxis.

Done... Hope it helps.

closed 2021-02-18 by lazybearcreations