RinCat/RTL88x2BU-Linux-Driver #77: [NO ISSUE] This also works for tp-link AC1300 Archer T4U PLus

htfy96, 2021-03-02:
Feel free to close this. Just want to help others find this repo after I spent the whole evening searching a working driver. Spamming some details below to increase its chance to be found.

FCC-ID: TE7T4UPlus. [It uses RTL 8812bu](https://fccid.io/TE7T4UPLUS/Internal-Photos/18-Internal-Photos-4924603).

Tested on Arch Linux with 5.11.2. No occasional freeze. Everything works great at least in client mode.

Logs:
```
kernel: usb 1-2: new high-speed USB device number 4 using xhci_hcd
kernel: usb 1-2: New USB device found, idVendor=2357, idProduct=0115, bcdDevice= 2.10
kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 1-2: Product: 802.11ac NIC
kernel: usb 1-2: Manufacturer: Realtek
kernel: usb 1-2: SerialNumber: 123456
kernel: RTW: module init start
kernel: RTW: rtl88x2bu v5.8.7.2_36899.20200819_COEX20191120-7777
kernel: RTW: build time: Mar  1 2021 22:06:36
kernel: RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
```

closed 2021-03-02 by htfy96

RinCat, 2021-03-02:
Ok, I added it to the support list. Thanks for report.

algeorge, 2022-04-23:
Thank you so much, I've been looking all over for information on the T4U Plus.