RinCat/RTL88x2BU-Linux-Driver #80: modprobe return error

rolf-nivels, 2021-03-30:
[Fedora 32]
[Kernel 5.11.10-100.fc32.x86_64]

Dear guys

After compiling with DKMS (or by using the make file) modprobe complains while loading:

Error x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000e033e470, val ffffffffc1a7abd3

Any ideas?

Best regards
Rolf

RinCat, 2021-03-30:
I have no problem using it under 5.11.10 in gentoo. Try check your kernel headers / dependent modules that match your kernel version.

rolf-nivels, 2021-03-30:
Hi RinCat

uname -r
`5.11.10-100.fc32.x86_64`

kernel-headers
`5.11.7-100.fc32`

These are the official releases downloaded via dnf.

Could this make such a difference?

Best regards
Rolf

RinCat, 2021-03-30:
You should never use mismatched kernel and headers. It will almost always cause problem.

rolf-nivels, 2021-03-30:
Hi RinCat

I've rebooted the system using 5.11.7-100, checked the all info, now they are identical. But the problem remains.

Any other ideas?

Best regards
Rolf

RinCat, 2021-03-30:
I can only suspect that there is something wrong in your system. Have you tried recompiling and reinstalling the driver?

rolf-nivels, 2021-03-30:
I tried it to compile via make, but the same result

rolf-nivels, 2021-03-30:
Compiling was without an messages, dmesg reports

module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000003c40d5ee, val ffffffffc1492bd3

rolf-nivels, 2021-03-30:
make.log from dkms:

`DKMS make.log for rtl88x2bu-git for kernel 5.11.7-100.fc32.x86_64 (x86_64)
Di 30 Mär 2021 15:59:07 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.11.7-100.fc32.x86_64/build M=/var/lib/dkms/rtl88x2bu/git/build  modules
make[1]: Verzeichnis „/usr/src/kernels/5.11.7-100.fc32.x86_64“ wird betreten
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_mcc.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/hal_hci/hal_usb.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/led/hal_led.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/led/hal_usb_led.o
[... 74 lines cut ...]
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/halrf_psd.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/rtl8822b/halhwimg8822b_bb.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/rtl8822b/halhwimg8822b_mac.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/rtl8822b/halrf_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/rtl8822b/phydm_hal_api8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/rtl8822b/halhwimg8822b_rf.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/rtl8822b/halrf_iqk_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/halrf/rtl8822b/halrf_rfk_init_8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/rtl8822b/phydm_regconfig8822b.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/phydm/rtl8822b/phydm_rtl8822b.o
[... cut ...]
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/hal/btc/halbtc8822b2ant.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl88x2bu/git/build/88x2bu.o
  MODPOST /var/lib/dkms/rtl88x2bu/git/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl88x2bu/git/build/88x2bu.mod.o
  LD [M]  /var/lib/dkms/rtl88x2bu/git/build/88x2bu.ko
make[1]: Verzeichnis „/usr/src/kernels/5.11.7-100.fc32.x86_64“ wird verlassen
Signing /var/lib/dkms/rtl88x2bu/git/build/88x2bu.ko
`

RinCat, 2021-03-30:
The make log shows you are still using  5.11.7 headers.

rolf-nivels, 2021-03-30:
Like I mentioned before I switched the kernel to 5.11.7-100:

uname -r
`5.11.7-100.fc32.x86_64`

kernel-devel
`Name         : kernel-devel
Version      : 5.11.7
Release      : 100.fc32
Architecture : x86_64
`

kernel-headers
`Version      : 5.11.7
Release      : 100.fc32
Architecture : x86_64
`

RinCat, 2021-03-30:
Do you have multiple C compilers / versions?

rolf-nivels, 2021-03-31:
As far as I can see I have only one compiler

gcc --version 
`gcc (GCC) 10.2.1 20201125 (Red Hat 10.2.1-9)`

rolf-nivels, 2021-04-01:
I tested the same code now under Fedora 33 and the module can be loaded via modprobe. I have the following information:

FC32:
boot 5.11.7-100
kernel-devel   5.11.7-100
kernel-headers 5.11.7-100
gcc 10.2.1 20201125
make 4.2.1
WORKS: NO

FC33:
boot 5.11.10-200
kernel-devel 5.11.10-200
kernel-header 5.11.7-200
gcc 10.2.1 20201125
make 4.3
WORKS: YES

Any ideas?

rolf-nivels, 2021-04-01:
I finally found the problem:

modprobe -f 88x2bu

was reporting 'Key was rejected by service'

I'm using SecureBoot with a signing key. At a certain update of Fedora 32 the UEFI boot entry was updated and my machine was using the new "Fedora Boot Entry" (which has not my public key installed). After booting with the old "Fedora Boot Entry" the driver was loaded without any error.

Many thanks for your support

closed 2021-04-01 by RinCat