RinCat/RTL88x2BU-Linux-Driver #82: Cannot connect to 5G networks

slicingmelon, 2021-04-20:
Hi,

I am using a TP-link AC1300 dongle and I've managed to install the driver on kali linux (5.10.0-kali7-amd64). I've noticed that it does not see my 5GHz access point at all. It does see the 2.4GHz one though.
My 5G AP is on a openwrt router:
Channel 161
VHT 80
Security WPA2/WPA3 

I've changed the channel to 36 and now I am able to see the wireless network in network manager but when I try to connect it doesn't work or it crashes. 

Any suggestions are welcome, thank you.

RinCat, 2021-04-20:
You need post some logs like `dmesg`.
It appears to be incompatible with your local channel regulations, since channel 161 is not all regions available.

slicingmelon, 2021-04-21:
OK, so my AP is now on `channel 36`, `VHT 80`. I think I am able to see it only after I did `sudo modprobe 88x2bu vht_enable=2`
It prompts for password but it doesn't connect, I get prompted for password again, similar to when you enter a wrong password. My password is correct, I copy pasted it from windows to make sure I don't have a typo, I've also checked /etc/NetworkManager/system-connections/SSID to make sure the password is correct.

dmesg log:
```
[Wed Apr 21 05:54:00 2021] usb 3-1: new high-speed USB device number 2 using xhci_hcd
[Wed Apr 21 05:54:00 2021] usb 3-1: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 2.10
[Wed Apr 21 05:54:00 2021] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Apr 21 05:54:00 2021] usb 3-1: Product: 802.11ac NIC
[Wed Apr 21 05:54:00 2021] usb 3-1: Manufacturer: Realtek
[Wed Apr 21 05:54:00 2021] usb 3-1: SerialNumber: 123456
[Wed Apr 21 05:54:00 2021] RTW: [HALMAC]11692M
                           HALMAC_MAJOR_VER = 1
                           HALMAC_PROTOTYPE_VER = 6
                           HALMAC_MINOR_VER = 5
                           HALMAC_PATCH_VER = 6*
[Wed Apr 21 05:54:00 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Wed Apr 21 05:54:01 2021] RTW: HW EFUSE
[...]
[Wed Apr 21 05:54:01 2021] RTW: hal_com_config_channel_plan chplan:0x26
[Wed Apr 21 05:54:01 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Wed Apr 21 05:54:01 2021] RTW: WARN [HALMAC][WARN]power state unchange!!
[Wed Apr 21 05:54:01 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Wed Apr 21 05:54:01 2021] RTW: WARN rtw_halmac_poweron: Power state abnormal, try to recover...OK
[Wed Apr 21 05:54:01 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Wed Apr 21 05:54:01 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Wed Apr 21 05:54:01 2021] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] Registry's RF PATH:UNKNOWN
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R
[Wed Apr 21 05:54:01 2021] RTW: [RF_PATH] NumTotalRFPath:2
[Wed Apr 21 05:54:01 2021] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[Wed Apr 21 05:54:01 2021] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[Wed Apr 21 05:54:01 2021] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[Wed Apr 21 05:54:01 2021] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[Wed Apr 21 05:54:01 2021] RTW: rtw_ndev_init(wlan0) if1 mac_addr=28:ee:52:19:35:2f
[Wed Apr 21 05:54:01 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Wed Apr 21 05:54:02 2021] RTW: txpath=0x3, rxpath=0x3
[Wed Apr 21 05:54:02 2021] RTW: txpath_1ss:0x1, num:1
[Wed Apr 21 05:54:02 2021] RTW: txpath_2ss:0x3, num:2
[Wed Apr 21 05:54:02 2021] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Wed Apr 21 05:54:02 2021] RTW: wlan0- hw port(0) mac_addr =28:ee:52:19:35:2f
[Wed Apr 21 05:54:07 2021] RTW: nolinked power save enter
[Wed Apr 21 05:54:07 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Wed Apr 21 05:54:08 2021] RTW: txpath=0x3, rxpath=0x3
[Wed Apr 21 05:54:08 2021] RTW: txpath_1ss:0x1, num:1
[Wed Apr 21 05:54:08 2021] RTW: txpath_2ss:0x3, num:2
[Wed Apr 21 05:54:08 2021] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Wed Apr 21 05:54:08 2021] RTW: wlan0- hw port(0) mac_addr =28:ee:52:19:35:2f
[Wed Apr 21 05:54:08 2021] RTW: nolinked power save leave
[Wed Apr 21 05:54:13 2021] RTW: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[Wed Apr 21 05:54:13 2021] RTW: start auth
[Wed Apr 21 05:54:13 2021] RTW: auth success, start assoc
[Wed Apr 21 05:54:14 2021] RTW: WARN _beamforming_leave: c0:1f:1e:16:a0:fa is neither beamforming ee or er!!
[Wed Apr 21 05:54:18 2021] RTW: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[Wed Apr 21 05:54:18 2021] RTW: WARN _beamforming_leave: c0:1f:1e:16:a0:fa is neither beamforming ee or er!!
[Wed Apr 21 05:54:18 2021] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL

[Wed Apr 21 05:54:18 2021] RTW: WARN _beamforming_leave: c0:1f:1e:16:a0:fa is neither beamforming ee or er!!
[Wed Apr 21 05:54:23 2021] RTW: nolinked power save enter
[Wed Apr 21 05:54:23 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Wed Apr 21 05:54:24 2021] RTW: txpath=0x3, rxpath=0x3
[Wed Apr 21 05:54:24 2021] RTW: txpath_1ss:0x1, num:1
[Wed Apr 21 05:54:24 2021] RTW: txpath_2ss:0x3, num:2
[Wed Apr 21 05:54:24 2021] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Wed Apr 21 05:54:24 2021] RTW: wlan0- hw port(0) mac_addr =0e:88:35:04:32:99
[Wed Apr 21 05:54:24 2021] RTW: nolinked power save leave
[Wed Apr 21 05:54:29 2021] RTW: nolinked power save enter
[... 132 lines cut ...]
[Wed Apr 21 05:57:50 2021] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[Wed Apr 21 05:57:51 2021] RTW: txpath=0x3, rxpath=0x3
[Wed Apr 21 05:57:51 2021] RTW: txpath_1ss:0x1, num:1
[Wed Apr 21 05:57:51 2021] RTW: txpath_2ss:0x3, num:2
[Wed Apr 21 05:57:51 2021] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Wed Apr 21 05:57:51 2021] RTW: wlan0- hw port(0) mac_addr =28:ee:52:19:35:2f
[... cut ...]
[Wed Apr 21 05:58:03 2021] start_addr=(0x20000), end_addr=(0x40000), buffer_size=(0x20000), smp_number_max=(16384)
[Wed Apr 21 05:58:03 2021] RTW: wlan0- hw port(0) mac_addr =28:ee:52:19:35:2f
[Wed Apr 21 05:58:03 2021] RTW: nolinked power save leave
[Wed Apr 21 05:58:08 2021] RTW: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[Wed Apr 21 05:58:08 2021] RTW: start auth
[Wed Apr 21 05:58:08 2021] RTW: auth success, start assoc
[Wed Apr 21 05:58:08 2021] RTW: WARN _beamforming_leave: c0:1f:1e:16:a0:fa is neither beamforming ee or er!!
[Wed Apr 21 05:58:12 2021] RTW: nolinked power save enter
```

ahmed-hant, 2025-11-05:
@slicingmelon  Hello, it's been a long time, haha. I also don't see 5G networks, only in Kali Linux, but they do appear on other systems. Could you please tell me how you did it?