RinCat/RTL88x2BU-Linux-Driver #83: Slow download speed

NuromenPt, 2021-04-23:
Greetings and first of all thanks for your work. 

I have Ubuntu 20.04, the wireless USB is TP-Link Archer T3U Plus connected at a USB 3 port and I used Manual DKMS installation. 

Everything worked and I was able to download at max speed alowed by a 867Mbit/s connection.

Meanwhile Ubuntu has been updated and now the maximum download rate that I'm able to achieve is 250/270Mbits/s (I used iperf3 to run the tests using my Asus router as server) 

Kernel is 5.8.0-50-generic

Intelnal Wi-Fi disconnected.

Help please.

RinCat, 2021-04-23:
Do you have a list what upgraded?

NuromenPt, 2021-04-24:
> Do you have a list what upgraded?

Hope this helps

https://pastebin.ubuntu.com/p/gZ5N8ZyQ2q/

NuromenPt, 2021-04-25:
Found this

lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    | Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtsx_usb, 480M
    | Port 2: Dev 3, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    | Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        | Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        | Port 3: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
        | Port 3: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
        | Port 4: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
        | Port 4: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M

Archer T3U Plus is connected to the same USB 3 port as before, but for some reason now it's recognized as connected to a USB 2 port.

I run again modprobe 88x2bu rtw_switch_usb_mode=1

Nothing changed :-/

NuromenPt, 2021-04-25:
Found a workaround. 

Following this post https://askubuntu.com/questions/1189587/usb-3-0-and-3-1-are-recognized-as-2-0

I created a file called 88x2bu.conf in /etc/modprobe.d/

In that file I putted

options 88x2bu rtw_switch_usb_mode=1

And rebooted the computer

Now the USB wifi is recognized as USB3 and I have full speed again. 

lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    | Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    | Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=rtsx_usb, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    | Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        | Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        | Port 3: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
        | Port 3: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
        | Port 4: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
        | Port 4: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M

splitcells, 2021-04-26:
@RinCat Does it make sense to add info about the /etc/modprobe.d/88x2bu.conf into the README?

RinCat, 2021-04-26:
@splitcells I prefer that users should try command first, rather than writing it into the default configuration.

NuromenPt, 2021-04-26:
I just don't understand why

sudo modprobe 88x2bu rtw_switch_usb_mode=1

Does noting now.

RinCat, 2021-04-26:
@NuromenPt you need `sudo modprobe -r 88x2bu` first if it already loaded.

NuromenPt, 2021-04-26:
> @NuromenPt you need `sudo modprobe -r 88x2bu` first if it already loaded.

Didn't know that. 

So this could also be solved with

`sudo modprobe -r 88x2bu`

Followed by

'sudo modprobe 88x2bu rtw_switch_usb_mode=1'

I suggest adding this info in Readme.

Will close this thread

closed 2021-04-27 by NuromenPt