RinCat/RTL88x2BU-Linux-Driver #84: wpa_supplicantnl80211: kernel reports: Authentication algorithm number required

georgesimeo, 2021-04-28:
I'm running Ubunru 20.04.2 LTS  and using usb wifi adapter TP-Link Archer T4U V3 as a hot spot 
Connection is not stable and randomly is being lost within few hours intervals.

After each reboot and re-onnection I'm getting these message in logs
wpa_supplicantnl80211: kernel reports: Authentication algorithm number required

dkms status
rtl88x2bu, git, 5.8.0-43-generic, x86_64: installed

RinCat, 2021-04-28:
Need more logs.

georgesimeo, 2021-04-29:
пр 29 02:54:08 NGC systemd-udevd[382]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
апр 29 02:54:08 NGC systemd-udevd[382]: Could not set AlternativeName= or apply AlternativeNamesPolicy= on eno1: File exists
апр 29 02:54:08 NGC systemd-udevd[382]: eno1: Could not apply link config, ignoring: File exists
апр 29 02:54:08 NGC kernel: AVX2 version of gcm_enc/dec engaged.
апр 29 02:54:08 NGC kernel: AES CTR mode by8 optimization enabled
апр 29 02:54:08 NGC kernel: at24 0-0050: supply vcc not found, using dummy regulator
апр 29 02:54:08 NGC kernel: at24 0-0050: 256 byte spd EEPROM, read-only
апр 29 02:54:08 NGC kernel: at24 0-0051: supply vcc not found, using dummy regulator
апр 29 02:54:08 NGC kernel: at24 0-0051: 256 byte spd EEPROM, read-only
апр 29 02:54:08 NGC kernel: at24 0-0052: supply vcc not found, using dummy regulator
апр 29 02:54:08 NGC kernel: at24 0-0052: 256 byte spd EEPROM, read-only
апр 29 02:54:08 NGC kernel: at24 0-0053: supply vcc not found, using dummy regulator
апр 29 02:54:08 NGC kernel: at24 0-0053: 256 byte spd EEPROM, read-only
апр 29 02:54:08 NGC systemd-udevd[379]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
апр 29 02:54:09 NGC mtp-probe[396]: checking bus 3, device 2: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-7"
апр 29 02:54:09 NGC mtp-probe[395]: checking bus 3, device 4: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-12"
апр 29 02:54:09 NGC mtp-probe[413]: checking bus 3, device 3: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-8"
апр 29 02:54:09 NGC mtp-probe[396]: bus: 3, device: 2 was not an MTP device
апр 29 02:54:09 NGC mtp-probe[413]: bus: 3, device: 3 was not an MTP device
апр 29 02:54:09 NGC mtp-probe[395]: bus: 3, device: 4 was not an MTP device
апр 29 02:54:09 NGC systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
апр 29 02:54:09 NGC kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
апр 29 02:54:09 NGC kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
апр 29 02:54:09 NGC kernel: 88x2bu: loading out-of-tree module taints kernel.
апр 29 02:54:09 NGC kernel: 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
апр 29 02:54:09 NGC kernel: RTW: module init start
апр 29 02:54:09 NGC kernel: RTW: rtl88x2bu v5.8.7.2_36899.20200819_COEX20191120-7777
апр 29 02:54:09 NGC kernel: RTW: build time: Apr 16 2021 03:37:26
апр 29 02:54:09 NGC kernel: RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
апр 29 02:54:09 NGC kernel: RTW: [HALMAC]11692M
                               HALMAC_MAJOR_VER = 1
                               HALMAC_PROTOTYPE_VER = 6
                               HALMAC_MINOR_VER = 5
                               HALMAC_PATCH_VER = 6*
апр 29 02:54:09 NGC kernel: RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
апр 29 02:54:09 NGC kernel: RTW: HW EFUSE
апр 29 02:54:09 NGC kernel: RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
апр 29 02:54:09 NGC kernel: RTW: 0x010: 32 32 33 33  33 33 2C 2C  2D 2D 2D 13  EE EE FF FF  
апр 29 02:54:09 NGC kernel: RTW: 0x020: FF FF 19 18  17 17 17 16  14 13 12 10  14 12 12 12  
апр 29 02:54:09 NGC kernel: RTW: 0x030: 13 EE FF FF  00 FF 0C E0  FF FF 31 31  31 31 31 31  
апр 29 02:54:09 NGC kernel: RTW: 0x040: 2B 2B 2B 2B  2B 13 EE 00  FF FF FF FF  1D 1C 1B 1A  
апр 29 02:54:09 NGC kernel: RTW: 0x050: 16 15 13 12  10 0E 11 0F  0F 0F 13 EE  FF FF 00 FF  
апр 29 02:54:09 NGC kernel: RTW: 0x060: 0C E0 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
апр 29 02:54:09 NGC kernel: RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
апр 29 02:54:09 NGC kernel: RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 ESCOD
апр 29 02:54:08 NGC kernel: RAPL PMU: hw unit of domain package 2^-14 Joules
апр 29 02:54:08 NGC kernel: RAPL PMU: hw unit of domain dram 2^-14 Joules
апр 29 02:54:08 NGC kernel: dcdbas dcdbas: Dell Systems Management Base Driver (version 5.6.0-3.4)
апр 29 02:54:08 NGC systemd-udevd[382]: Using default interface naming scheme 'v245'.
апр 29 02:54:08 NGC kernel: cryptd: max_cpu_qlen set to 1000

пр 29 02:54:09 NGC systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.
апр 29 02:54:09 NGC kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
апр 29 02:54:09 NGC kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
апр 29 02:54:09 NGC kernel: 88x2bu: loading out-of-tree module taints kernel.
апр 29 02:54:09 NGC kernel: 88x2bu: module verification failed: signature and/or required key missing - tainting kernel
апр 29 02:54:09 NGC kernel: RTW: module init start
апр 29 02:54:09 NGC kernel: RTW: rtl88x2bu v5.8.7.2_36899.20200819_COEX20191120-7777
апр 29 02:54:09 NGC kernel: RTW: build time: Apr 16 2021 03:37:26
апр 29 02:54:09 NGC kernel: RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
апр 29 02:54:09 NGC kernel: RTW: [HALMAC]11692M
                               HALMAC_MAJOR_VER = 1
                               HALMAC_PROTOTYPE_VER = 6
                               HALMAC_MINOR_VER = 5
                               HALMAC_PATCH_VER = 6*
апр 29 02:54:09 NGC kernel: RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
апр 29 02:54:09 NGC kernel: RTW: HW EFUSE
апр 29 02:54:09 NGC kernel: RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
апр 29 02:54:09 NGC kernel: RTW: 0x010: 32 32 33 33  33 33 2C 2C  2D 2D 2D 13  EE EE FF FF  
[... cut ...]
апр 29 02:54:10 NGC systemd[1]: Starting Load/Save RF Kill Switch Status...
апр 29 02:54:10 NGC systemd-udevd[380]: Using default interface naming scheme 'v245'.
апр 29 02:54:10 NGC kernel: RTW: rtw_ndev_init(wlan0) if1 mac_addr=50:3e:aa:bc:90:a3
апр 29 02:54:10 NGC kernel: usbcore: registered new interface driver rtl88x2bu
апр 29 02:54:10 NGC kernel: RTW: module init ret=0
апр 29 02:54:10 NGC systemd-udevd[380]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.

georgesimeo, 2021-04-29:
апр 29 02:54:20 NGC wpa_supplicant[712]: Successfully initialized wpa_supplicant
апр 29 02:54:24 NGC NetworkManager[670]: <info>  [1619654064.1493] supplicant: wpa_supplicant running
апр 29 02:54:24 NGC wpa_supplicant[712]: nl80211: kernel reports: Authentication algorithm number required
апр 29 02:54:27 NGC wpa_supplicant[712]: wlx503eaabc90a3: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
апр 29 02:54:27 NGC wpa_supplicant[712]: Note: nl80211 driver interface is not designed to be used with ap_scan=2; this can result in connection failures
апр 29 02:54:28 NGC wpa_supplicant[712]: Using interface wlx503eaabc90a3 with hwaddr 50:3e:aa:bc:90:a3 and ssid "NGC"
апр 29 02:54:28 NGC wpa_supplicant[712]: wlx503eaabc90a3: interface state UNINITIALIZED->ENABLED
апр 29 02:54:28 NGC wpa_supplicant[712]: wlx503eaabc90a3: AP-ENABLED
апр 29 02:54:28 NGC wpa_supplicant[712]: wlx503eaabc90a3: CTRL-EVENT-CONNECTED - Connection to 50:3e:aa:bc:90:a3 completed [id=0 id_str=]
апр 29 02:54:39 NGC wpa_supplicant[712]: wlx503eaabc90a3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
апр 29 02:54:39 NGC wpa_supplicant[712]: wlx503eaabc90a3: AP-STA-CONNECTED f8:a2:d6:5c:39:1d
апр 29 02:55:20 NGC wpa_supplicant[712]: wlx503eaabc90a3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
апр 29 02:55:20 NGC wpa_supplicant[712]: wlx503eaabc90a3: AP-STA-CONNECTED 00:eb:a0:27:36:86

RinCat, 2021-04-29:
https://w1.fi/cgit/hostap/plain/wpa_supplicant/wpa_supplicant.conf
Try set `ap_scan=1`?

adityamj, 2021-10-06:
I am facing the same issue on Manjaro
`
$uname -a
Linux sirius 5.14.7-2-MANJARO #1 SMP PREEMPT Wed Sep 22 12:22:56 UTC 2021 x86_64 GNU/Linux
`
Here are the logs https://pastebin.com/eRVVeQ9z

adityamj, 2021-10-06:
Thanks @RinCat ,
The AP was using WPA3-SAE.
I tried using WPA2-PSK with CCMP AES, which connects to the network.

RinCat, 2021-10-06:
@adityamj it should supported WPA3-SAE, but not GCMP.

adityamj, 2021-10-06:
I am using Openwrt, and on the dashboard if WPA3-SAE is chosen, it does not show an option to choose the algorithm.

[image]

It probably defaults to GCMP, but I have no ideal

RinCat, 2021-10-06:
It probably a bug somewhere in openwrt or wpa_supplicant or NM. 
WPA3-SAE should use CCMP-128.

RinCat, 2021-12-04:
You may try it again, the new driver should support GCMP.