RinCat/RTL88x2BU-Linux-Driver #87: armv7 support

yananet, 2021-05-14:
Hello,

Do you support armv7(Raspberry Pi)?
I tried on Arch Linux, but no success.
```
$ cat /var/lib/dkms/rtl88x2bu/5.8.7.1.r119.2ed61f8/build/make.log 
DKMS make.log for rtl88x2bu-5.8.7.1.r119.2ed61f8 for kernel 5.10.33-2-ARCH (armv7l)
Fri May 14 09:27:11 AM JST 2021
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.10.33-2-ARCH/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1.r119.2ed61f8/build  modules
make[1]: Entering directory '/usr/lib/modules/5.10.33-2-ARCH/build'
Makefile:682: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.10.33-2-ARCH/build'
make: *** [Makefile:2378: modules] Error 2
```

RinCat, 2021-05-14:
Try `make ARCH=arm` or `make ARCH=arm64`
https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/71

yananet, 2021-05-25:
Thank you!
DKMS successfully done, but not worked for me...
the interface doesn't show up.
```
[ 2745.774940] usb 1-1.5.1: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[ 2745.778452] usb 1-1.5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2745.780271] usb 1-1.5.1: Product: 802.11ac NIC
[ 2745.782075] usb 1-1.5.1: Manufacturer: Realtek
[ 2745.783911] usb 1-1.5.1: SerialNumber: 123456
[ 2745.930249] RTW: module init start
[ 2745.931142] RTW: rtl88x2bu v5.8.7.2_36899.20200819_COEX20191120-7777
[ 2745.931999] RTW: build time: May 25 2021 11:44:26
[ 2745.932835] RTW: rtl88x2bu BT-Coex version = COEX20191120-7777
[ 2745.934921] RTW: [HALMAC]11692M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 5
               HALMAC_PATCH_VER = 6*
[ 2745.940143] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2746.366219] RTW: HW EFUSE
[ 2746.367182] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00  
[ 2746.368177] RTW: 0x010: 32 32 32 32  32 32 30 30  30 30 30 05  00 00 FF FF  
[ 2746.369145] RTW: 0x020: FF FF 2E 2E  2D 2D 2D 2D  2D 2D 2D 2D  2D 2D 2D 2D  
[ 2746.370100] RTW: 0x030: 02 00 FF FF  00 FF EC 00  FF FF 2D 2D  2D 2D 2D 2D  
[ 2746.371046] RTW: 0x040: 2D 2D 2D 2D  2D 02 00 00  FF FF FF FF  2A 2A 2C 2C  
[ 2746.371975] RTW: 0x050: 2C 2C 2C 2C  2C 2C 2D 2D  2D 2D 22 00  FF FF 00 FF  
[ 2746.372892] RTW: 0x060: 3C 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.373838] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.374731] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.375586] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.376391] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.377158] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 28 25 00  FF FF FF FF  
[ 2746.377901] RTW: 0x0C0: FF 01 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF  
[ 2746.378632] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.379339] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.380009] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.380654] RTW: 0x100: DA 0B 12 B8  E1 67 02 00  13 EF F2 1B  B2 09 03 52  
[ 2746.381316] RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
[ 2746.381993] RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
[ 2746.382679] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.383394] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF  
[ 2746.384103] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.384789] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.385379] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.385874] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.386354] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.386833] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.387306] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.387763] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.388205] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.388627] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.389031] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.389417] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.389790] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.390149] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.390491] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.390837] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.391177] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.391502] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.391825] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.392148] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.392466] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.392771] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.393065] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.393358] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.393652] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.393950] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.394218] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[ 2746.394473] RTW: hal_com_config_channel_plan chplan:0x7F
[ 2746.394685] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2746.407434] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[ 2746.511577] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2746.512132] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2746.512628] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[ 2746.513161] RTW: [RF_PATH] ver_id.RF_TYPE:RF_2T2R, rf_reg_path_num:2, max_tx_cnt:2
[ 2746.513494] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[ 2746.513876] RTW: [RF_PATH] Registry's RF PATH:UNKNOWN
[ 2746.514233] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x33, max_tx_cnt:2
[ 2746.514613] RTW: [RF_PATH] HALDATA's rf_type:RF_2T2R
[ 2746.514995] RTW: [RF_PATH] NumTotalRFPath:2
[ 2746.515385] RTW: [TRX_Nss] HALSPEC - tx_nss :2, rx_nss:2
[ 2746.515797] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[ 2746.516221] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:2
[ 2746.517730] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[ 2746.523062] RTW: rtw_ndev_init(wlan0) if1 mac_addr=00:13:ef:f2:1b:b2
[ 2746.528427] usbcore: registered new interface driver rtl88x2bu
```

yananet, 2021-05-25:
Sorry, It worked successfully after a reboot, thanks again!

closed 2021-05-25 by yananet